python-control/python-control #236: AttributeError: module 'control' has no attribute 'matlab'

CPCLAB-UNIPI, 2018-09-20:
Hi, 
I am trying to simulate the output of a linear system using lsim and more specifically control.matlab.lsim . 
I have installed control 0.8.0 on Py3.6 trough pip command and the installation result successful. When I write the command control.lsim it gives me the following error 
- AttributeError: module 'control' has no attribute 'lsim'

Then when I modify the command into control.matlab.lsim as written in the documentation found here (https://python-control.readthedocs.io/en/0.8.0/generated/control.matlab.lsim.html), I get this other error:
- AttributeError: module 'control' has no attribute 'matlab'

I already tried to uninstall the package control and reinstall it but the problem is still there. 
Since I am using Anaconda I also checked if there is a conflict with a package control installed via conda but the only package installed is the one via pip and it is the 0.8.0. 

I also know other users had the same problem with the same command. 
How can I solve this problem? 

Thanks

CPCLAB-UNIPI, 2018-09-20:
import control as cnt
err, t, x = cnt.lsim(sys, err, t)

where sys is a Transfer function obtained trough the command cnt.tf 

Anyway the control package that I have installed has no 'lsim' attribute nor 'matlab' attribute. And I have installed it with "pip".

murrayrm, 2018-09-20:
The error is coming because `lsim` is not a command in the `control` package.  You have to import `control.matlab` if you want to use the MATLAB-like interface:
```
import control.matlab as cnt
import numpy as np
sys = cnt.tf([1], [1,1])
u = np.sin(T)
T = cnt.linspace(0,10,100)
(yout, Tout, xout) = cnt.lsim(sys, u, T)
```

CPCLAB-UNIPI, 2018-09-20:
Ok, thanks for the answer. This solves a big misunderstand around the control package.

closed 2018-09-21 by murrayrm