python-control/python-control #237: matlab is not include in __init__.py

murrayrm, 2018-09-20:
Following up on issue #236, the following code seems like it should work but doesn't:
```
import control
import numpy as np
sys = control.tf([1], [1,1])
u = np.sin(T)
T = control.matlab.linspace(0,10,100)
(yout, Tout, xout) = cnt.matlab.lsim(sys, u, T)
```
The problem is that the `matlab` module is not actually imported in `control/__init__.py` and so it has to be imported separately.

moorepants, 2018-09-20:
In general, a python user shouldn't rely on super modules importing sub modules into the namespace. You can certainly set things up that way, but it is probably best to decide exactly what you want in the top level namspace. The user is generally responsible for importing any pacakge or module that they want to use functions, variables, and classes from. The user fix for something like this is:

```
In [3]: import control

In [4]: control.matlab.linspace(0, 10, 100)
---------------------------------------------------------------------
AttributeError                      Traceback (most recent call last)
<ipython-input-4-f7e2e3c31a7d> in <module>()
----> 1 control.matlab.linspace(0, 10, 100)

AttributeError: module 'control' has no attribute 'matlab'

In [5]: import control.matlab

In [6]: control.matlab.linspace(0, 10, 100)
Out[6]: 
array([  0.        ,   0.1010101 ,   0.2020202 ,   0.3030303 ,
         0.4040404 ,   0.50505051,   0.60606061,   0.70707071,
         0.80808081,   0.90909091,   1.01010101,   1.11111111,
         1.21212121,   1.31313131,   1.41414141,   1.51515152,
         1.61616162,   1.71717172,   1.81818182,   1.91919192,
         2.02020202,   2.12121212,   2.22222222,   2.32323232,
         2.42424242,   2.52525253,   2.62626263,   2.72727273,
         2.82828283,   2.92929293,   3.03030303,   3.13131313,
         3.23232323,   3.33333333,   3.43434343,   3.53535354,
         3.63636364,   3.73737374,   3.83838384,   3.93939394,
         4.04040404,   4.14141414,   4.24242424,   4.34343434,
         4.44444444,   4.54545455,   4.64646465,   4.74747475,
         4.84848485,   4.94949495,   5.05050505,   5.15151515,
         5.25252525,   5.35353535,   5.45454545,   5.55555556,
         5.65656566,   5.75757576,   5.85858586,   5.95959596,
         6.06060606,   6.16161616,   6.26262626,   6.36363636,
         6.46464646,   6.56565657,   6.66666667,   6.76767677,
         6.86868687,   6.96969697,   7.07070707,   7.17171717,
         7.27272727,   7.37373737,   7.47474747,   7.57575758,
         7.67676768,   7.77777778,   7.87878788,   7.97979798,
         8.08080808,   8.18181818,   8.28282828,   8.38383838,
         8.48484848,   8.58585859,   8.68686869,   8.78787879,
         8.88888889,   8.98989899,   9.09090909,   9.19191919,
         9.29292929,   9.39393939,   9.49494949,   9.5959596 ,
         9.6969697 ,   9.7979798 ,   9.8989899 ,  10.        ])
```

It is also not good use practice importing numpy.linspace from control.matlab, better practice is importing numpy functions directly from numpy. If control.matlab had `from numpy import linspace as _linspace` instead of `from numpy import linspace` in its `__init__.py` file that would help discourage this behavior. It also relieves control from supporting numpy functions as part of its API.

murrayrm, 2018-09-20:
My thought was just that since `matlab` is a submodule of `control`, it should at least be accessible as `control.matlab` (like `scipy.signal`, which you can get at just by doing `import scipy`).

In terms of importing `numpy` function: agree in principle.  The idea in the `matlab` submodule is to allow users familiar with MATLAB to use `from control.matlab import *` and then type the "usual" commands.

Would be interested in feedback on this approach (from @moorepants and others) so that we do something that feels rational.

moorepants, 2018-09-20:
> My thought was just that since matlab is a submodule of control, it should at least be accessible as control.matlab (like scipy.signal, which you can get at just by doing import scipy).

It's fine to set it up this way. I recommend being consistent with all modules/subpackages. Note that scipy doesn't generally allow this:

```
In [1]: import scipy

In [2]: scipy.optimize.minimize
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-2-ec8b0b5e3bc3> in <module>()
----> 1 scipy.optimize.minimize

AttributeError: module 'scipy' has no attribute 'optimize'

In [3]: scipy.integrate.trapz
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-abd5c443b6df> in <module>()
----> 1 scipy.integrate.trapz

AttributeError: module 'scipy' has no attribute 'integrate'

In [4]: scipy.signal.convolve
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-4-ef600dca46b2> in <module>()
----> 1 scipy.signal.convolve

AttributeError: module 'scipy' has no attribute 'signal'

In [5]: import scipy.optimize

In [6]: import scipy.integrate

In [7]: import scipy.signal

In [8]: scipy.optimize.minimize
Out[8]: <function scipy.optimize._minimize.minimize(fun, x0, args=(), method=None, jac=None, hess=None, hessp=None, bounds=None, constraints=(), tol=None, callback=None, options=None)>

In [10]: scipy.integrate.trapz
Out[10]: <function numpy.lib.function_base.trapz(y, x=None, dx=1.0, axis=-1)>

In [11]: scipy.integrate.trapz
Out[11]: <function numpy.lib.function_base.trapz(y, x=None, dx=1.0, axis=-1)>
```

> In terms of importing numpy function: agree in principle. The idea in the matlab submodule is to allow users familiar with MATLAB to use from control.matlab import * and then type the "usual" commands.

That seems to be a valid reason to allow matching numpy/matlab functions to be imported from the matlab module. Much like matplotlib's pylab module. But my hunch is that matplotlib would remove that if they could...

murrayrm, 2018-12-23:
Took another look at this and I think it makes sense to leave things as they are.  If you want to use MATLAB commands then you need to import `control.matlab`.

closed 2018-12-23 by murrayrm