python-control/python-control #238: Add Delay in Expoenential Form

Vitorbnc, 2018-10-02:
First, great library! It just works out of the box and has a lot of functions one would expect when dealing with LTI systems.

My teacher's assignment asks me to add a delay to a system multiplying it by `exp(-0.5*s)` , but I found no way of doing it in python-control. There's the `pade` function, but that's an approximation. Is it currently possible?

murrayrm, 2018-10-02:
Pure time delays are not currently implemented.  If you would like to contribute to the package, it would be interesting to see an implementation of them (non-trivial, since it would probably require substantial changes to the various objects and operations).

technip-energies-cma, 2021-02-02:
Hi,

I also think that this library is really a good job. I will try to add this pure delay functionality. Hope is it not too difficult !

Camille

murrayrm, 2021-12-16:
Duplicates #628, where there is a bit more extensive discussion.

closed 2021-12-16 by murrayrm