python-control/python-control #239: non-intuitive Time-Domain Simulation functions behavior

hungpham2511, 2018-10-25:
Hi, I notice that the simulation functions: `step_response`, `impulse_response`, for discrete time systems is not very intuitive, and is also inconsistent with matlab's behavior (I sort of have the habit of checking matlab documentation for a python-control function if there are suprises). 

Consider this snippet
```python
import control as co
import numpy as np
A1 = co.tf([1], [1, 6, 5])
A1 = co.c2d(A1, 0.5)
T, y = co.step_response(A1, np.arange(0, 5, 0.2))
```

we would expect `T` to be the time instances that correspond to values in `y` (matlab behavior), but it turns out not the case. I got `T.shape = (25,)` while `y.shape=(1, 10)`.  The same behavior can be observed for other time-domain functions for discrete-time systems (continuous-time systems are alright).

ilayn, 2018-10-25:
It is just giving you back the time array but since the sampling time of the system doesn't match the time increment it is resampling the bounds.

hungpham2511, 2018-11-22:
That's right. 

However, this behavior is not intuitive to users. I would prefer it to be the actual time-array that is used to compute the impulse response. This information can be used, say, for plotting. Also, with the current behavior, the returned time-array T does not convey any information.

murrayrm, 2018-12-28:
I agree that this behavior is non-intuitive and probably incorrect.  Marking as a bug to be fixed in the next release.

murrayrm, 2019-01-05:
I did a bit of looking into this.  Some observations and issues:

* The 'bug' in this code is something that is apparently inherited from the `scipy.signal.dlsim()`, which is what `step_response()` eventually calls (through a call to `forced_response()`).  In particular, it appears that if you pass `dlsim()` a time vector that doesn't match the sampling time, it doesn't generate an error and, furthermore, it returns whatever you passed it (same behavior as what was noticed here).

* There is a further inconsistency in the way that `T=None` is handled in the `control` simulation functions for discrete time systems: the time vector that is generated will have `dt=1` independent of what the actually sampling time for the system is.

* There may also be some issues with the way that the default response time is determined when you pass `T=None` for discrete time systems.  For `step_response()` and `initial_response()` this is supposed to be based on the dynamics matrix `A`.  This uses the `_default_response_times()` function from `scipy.signal.ltisys`, but I believe that function assumes continuous time (need to check).

I think that the correct behavior for discrete time systems is the following (which would match MATLAB):

* The values of the output argument `T` should always match the times at which the corresponding outputs `Y` are obtained.  This is consistent with the idea that you plot the outputs of a simulation function by using commands of the form
```
t, y = step_response(sys)
plot(t, y)
```

* If you pass an explicit set of sampling times for a discrete time system, the time increment should match the sampling time (with `dt=True` being interpreted as a sampling time of 1).  In this case, the output argument `T` will match the input argument `T`.

* If you omit the sampling time, the value of `T` will be determined based on either the input signal (for `forced_response()` or the time constants coming from the system dynamics.  In either case, the difference between time points should be equal to the sampling time of the system (or 1 if `dt=True`).

* For `forced_response()`, the length of the input vector `U` should match the length of the time vector `T`.

The good news is that a relatively modest set of changes are required to fix all of these things up.  Comments welcome on whether I am missing anything, but I'll start putting together test cases for the properties above and then updating the code.

murrayrm, 2019-04-11:
Fixed in PR #286

murrayrm, 2019-04-20:
Will look into this and see if I can generate a unit test that exhibits the error, so that we can nail this one down.

murrayrm, 2019-04-21:
This behavior has been fixed in PR #295.

murrayrm, 2019-05-02:
PR #295 is merged.  Closing this (again).

closed 2019-05-02 by murrayrm