python-control/python-control #243: MacOS version of python-control crashes with matplotlib v3.0

murrayrm, 2018-11-05:
While debugging pull request #242, I was getting crashes on MacOS 10.13 (High Sierra).  Finally chased the problem down to `matplotlib-3.0.1`, which was having some problems with the backend.  Documenting this here so that we can see if it is an issue for others or on other platforms.

Quick fix: set the `MPLBACKEND` environment variable to `TkAgg`
```
export MPLBACKEND="TkAgg" 
```
Alternative: use matplotlib-2.2.2.

kangwonlee, 2018-12-06:
May I have your opinion on a multi-platfrom version of the `.travis.yml` for both linux and osx?  Would be helpful even by a little bit?

murrayrm, 2018-12-28:
@kangwonlee Comment above is only peripherally aligned with this issue.   I am going to delete the comment from here and suggest that you create a new issue with your comments so that we can discuss further.

murrayrm, 2019-04-12:
This problem seems to be gone on MacOS 10.14.3 (Mojave) and matplotlib-3.0.0 under python 3.5.

murrayrm, 2019-04-20:
Removing `wontfix`.  Looks like this is still a problem on the conda-forge build for MacOS.

murrayrm, 2020-12-28:
This issue seems to have been fixed.  Just tested with python3.7, control-0.8.4, matplotlib-3.32 on Big Sur and all worked fine.

closed 2020-12-28 by murrayrm