python-control/python-control #249: Conda v0.8.0 missing lsim

urogers, 2018-12-13:
Feel this might be a Anaconda issue, but updating to the new control 0.8.0 package on Windows, the lsim module appears to be missing.  

Reading The Docs:  It clearly shows the lsim MATLAB compatibility module as part of the package.

This was testing on a Windows 7 and Windows 10 Install.  Reverting back to 0.7.0 corrects the issue.

murrayrm, 2018-12-22:
Can you send a code snippet just to make sure we are looking at the same thing.  I did a conda install on MacOS 10.14.2 with Python 3.5 using the command
```
conda install -c conda-forge control=0.8.0
```
and the following works
```
from control.matlab import *
sys = tf([1], [1,1])
yout, T, xout = lsim(sys, 0, linspace(0,10), 0)
```
(not a very interesting simulation, but I don't get an error).  If you get an error with the same code on Windows, could be a problem in our conda package.

urogers, 2018-12-22:
The following run on 0.7.0, but fails on 0.8.0.  The import control * appears to pull in lsim from the MATLAB subpackage on 0.7.0, but not on 0.8.0.  The linspace fails in both versions.  Using 'from control.matlab import *', works with no error.

`from control import *
import numpy as np
sys = tf([1], [1,1])
yout, T, xout = lsim(sys, 0, np.arange(0,1,0.01), 0)`

Given this result, it is straightforward to update my legacy code to pull in the MATLAB subpackage when it is needed.

murrayrm, 2018-12-23:
Indeed, `lsim()` is not part of the `control` package, but part of the MATLAB compatibility subpackage, `control.matlab`.  So this is the intended behavior.  See issue #237 for some discussion around this.

Note that in the `control` package, the equivalent function is `forced_response()`.

closed 2018-12-23 by murrayrm