python-control/python-control #256: Create functions for better indexing of MIMO transfer functions

murrayrm, 2018-12-28:
Motivated by PR #246, which implemented indexing for transfer functions, it would be nice to be able to use Python slice notation for MIMO system objects.  So if you had a MIMO transfer function G, you could say things like G[:,1:3] to get a subset of the transfer functions.

A full implementation should allow slide functionality to be used in simulation and plotting functions, block diagram functions (issue #20), etc.  

Will use this issue to keep track of some of the possibilities and see if there is a good way forward.

roryyorke, 2018-12-28:
What is "slide functionality" ?

I glanced at #20, it's not immediately obvious to me how this is related to MIMO slicing, other than allowing a MIMO sub-object to be submitted for block diagram operations.

If I read #246 right, it does implement this for TFs - do you want this for FRD and SS objects too?  Maybe change the title to "MIMO system objects", as it is in the text.

roryyorke, 2018-12-28:
Ah, right; I think my mind jumped to GUI sliders.

Unfortunately, and unlike Matlab, Python only supports slice (and ellipsis ```...```) notation when indexing (using ```[]```), so ```foo(1:2)``` is a syntax error.  One could allow ```feedback(f, g, slice(1,2,3), slice(4,5,6))``` - is that what you had in mind?

Tangentially related: the last time I used the control toolbox, I found the signal naming capability very handy; this lets one combine a number of LTI objects using ```connect```; assuming all the objects have the correct names, one only has to specify the final inputs and outputs, as cell arrays of strings, and the toolbox does the rest.

rega0051, 2021-01-06:
@roryyorke I use signal naming extensively. I wrote a little wrapper around 'connect' for my work. Would like to have I/O names as an element of the systems so they are preserved through normal transformations. As it is I use the ConnectName function for adding, multiplying, feedback operations; but its not very clean. Slicing should also be accomplished by name, rather than trying to keep track of the indices. I've always found managing named signals much less error prone than counting out index numbers.

'''
# Controller definition
sysK = control.ss([], [], [], 0.5 * np.eye(2))
sysK.InputName = ['e1', 'e2']
sysK.OutputName = ['uCtrl1', 'uCtrl2']

# Reference Input: e = r - z
sysSumR = control.ss([], [], [], [[1, 0, -1, 0], [0, 1, 0, -1]])
sysSumR.InputName = ['r1', 'r2', 'z1', 'z2']
sysSumR.OutputName = ['e1', 'e2']

connectNames = sysK.InputName
inKeep = sysSumR.InputName
outKeep = sysK.OutputName
sysCtrl = ConnectName([sysSumR, sysK], connectNames, inKeep, outKeep)

'''

```
def ConnectName(sysList, connectNames, inKeep, outKeep):

    sys = []
    for s in sysList:
        if sys == []:
            sys = s
        else:
            sys = control.append(sys, control.ss(s))

    inNames = []
    outNames = []
    for s in sysList:
        inNames += s.InputName
        outNames += s.OutputName

    Q = [[inNames.index(s)+1, outNames.index(s)+1]  for s in connectNames]

    inputv = [inNames.index(s)+1 for s in inKeep]
    outputv = [outNames.index(s)+1 for s in outKeep]

    sysOut = control.connect(sys, Q, inputv, outputv)

    sysOut.InputName = inKeep
    sysOut.OutputName = outKeep

    return sysOut

```

murrayrm, 2021-01-06:
FYI, the `iosys` module supports signal naming and there is an `interconnect` function the latest master that supports this sort of named connection structure. See #497 for some details.

rega0051, 2021-01-08:
@murrayrm  Thanks, I'll look for it.

ilayn, 2021-01-09:
It is actually not that sophisticated. It's the slice notation of numpy and python that is a bit annoying. 

https://github.com/ilayn/harold/blob/master/harold/_classes.py#L903-L928

Same is also present for the State class in the same file

ilayn, 2021-01-09:
oops just saw the timestamps sorry for the noise.

closed 2024-06-27 by murrayrm