python-control/python-control #257: Travis-CI support for python on Linux and OSX

kangwonlee, 2018-12-28:
* May I have your opinion on a multi-platfrom version of the `.travis.yml` for both `linux` and `osx`?  Would be helpful even by a little bit?
* I just [noticed](https://travis-ci.org/kangwonlee/python-control/builds/464316186) that the [Travis-CI](https://docs.travis-ci.com/user/reference/osx/) support for python in `OSX` seems a little less positive according to this discussion : https://github.com/travis-ci/travis-ci/issues/9744#issuecomment-419426053
* IMHO, it would be possible to support both `linux` and `osx`; however, the `.travis.yml` may look like a [spaghetti code](https://github.com/kangwonlee/python-control/blob/travis-osx/.travis.yml) 
* If Travis-CI could support multi-layer [`env` `matrix`](https://github.com/travis-ci/travis-ci/issues/1519), the code could be more compact, however, no plan yet. https://github.com/travis-ci/travis-ci/issues/1519#issuecomment-124146725
* Also, the during the `osx`tests, `test_discrete_bode()` is [terminating](https://travis-ci.org/kangwonlee/python-control/jobs/464355186#L550) due to `NSInvalidArgumentException`.  Is this the same type of crash you mentioned above? (I could see that the [differences](https://travis-ci.org/kangwonlee/python-control/builds/464369087) with the `MPLBACKEND="TkAgg"` line.)

murrayrm, 2018-12-28:
It could be useful to do unit tests on MacOS (and Windows), but it would mean that unit tests on commits will take longer to run (your build above has a run time of 1 hr 6 min, compared with ~36 min for current test framework).  For me, I think testing under linux is enough and we can rely on local testing by developers to make sure that Windows and MacOS versions continue to work.

I would be interested in hearing from others about whether they think the extra testing would be useful.  

In response to your last bullet regarding the failure you observed: the line you highlighted is due to the issue identified in #243.  I notice that there are also problems  with `slycot` not compiling properly (this appears in the overall build).

murrayrm, 2019-12-29:
This issue is a year old => stale.  Closing.

closed 2019-12-29 by murrayrm