python-control/python-control #272: ss2tf and tf(sys) create the wrong results

jzilly, 2019-01-08:
The transfer function of 
A = np.array([[-4, -1],[-1 , -4]])
B = np.array([[1],[3]])
C = np.array([[3,1]])
D = 0
sys = control.ss2tf(A, B, C, D)
is calculated to be (6*s+13)/(s^2+8*s+15)

However, the correct answer is (6*s+14)/(s^2+8*s+15) - (14 instead of 13 in nominator).

This happens also for other transfer functions of state space matrices making this function unusable in practice. Please fix.

don4get, 2019-01-08:
Indeed, it only affect python 2.7.
At least, it worked for me on python 3.4.9.
I added unit tests on this branch:  https://github.com/python-control/python-control/pull/273
As explained here, the problem comes from the fact that in python to division between ints returns an int.

The easy fix would be to write your system as follows:
```A = np.array([[-4., -1.],[-1. , -4.]])
B = np.array([[1.],[3.]])
C = np.array([[3.,1.]])
D = 0.
sys = control.ss2tf(A, B, C, D)```

jzilly, 2019-01-08:
It also affects this code that I ran using Python3 in a Jupyter notebook. Converting the values to float however fixes the issue. Can you convert all values to float by default?

murrayrm, 2019-04-07:
Fixed in  PR  #273

closed 2019-04-07 by murrayrm