python-control/python-control #275: Issue with discrete-time transfer function simulation for MIMO system

franciscofernandezf, 2019-01-30:
I have an issue trying to simulate a discrete-time transfer function for a MIMO system, using the function `forced_response()`. I used the `lsim()` function in MATLAB to compare and I got a correct response, but when I tried it in Python, I get a vector of zeros as a response followed by these messages: 

```
Wrong parameter 1 in LAPACKE_chetrs_rook
Wrong parameter 1 in LAPACKE_dsyev_2stage_work
Wrong parameter 1 in LAPACKE_chbevd_2stage_work
Wrong parameter 1 in LAPACKE_dsyev_2stage_work
Wrong parameter 1 in LAPACKE_dsytrs
Wrong parameter 1 in LAPACKE_cheswapr_work
Wrong parameter 1 in LAPACKE_cgetrs
Wrong parameter 1 in LAPACKE_cgtsv_work
Wrong parameter 1 in LAPACKE_cgetrf_work
```
I tried using a code with a different function that I found on the internet and I get the same problem, discarding the coding as the cause.

I am running Windows 10 PRO 64 bits, Python 2.7 64 bits, PyCharm, control-0.8.1, scipy-1.2.0. 
I installed numpy+mkl and slycot from the binnary wheels I found on https://www.lfd.uci.edu/~gohlke/pythonlibs/ using pip-19.0.1.

murrayrm, 2019-02-01:
Can you include some example code that generates the error message?

franciscofernandezf, 2019-02-11:
I cannot send you my code, but using the code from issue #113 

```
import control as ctl
import control.matlab as ml
import numpy as np
import matplotlib.pyplot as plt

j = np.complex(0,1)
a = -7+j*np.sqrt(51)
b = np.conjugate(a)

# Sistema en contínuo
sys1c_num, sys1c_den = ml.zpk2tf([], [-.5, a, b], [-100])
sys1c = ctl.TransferFunction(sys1c_num, sys1c_den)
Tc, y1c = ctl.impulse_response(sys1c)
plt.plot(Tc, y1c, label='continuous');

# Sistema Discreto

dt = 0.001
sys1d = ctl.sample_system(sys1c, dt, method='zoh')

td = np.arange(0.,Tc[-1]+dt,dt);
u1d = np.zeros(td.shape);
u1d[0] = 1/dt # seems to correspond to a continuous time impulse

Td, y1d, _ = ctl.forced_response(sys1d, td, u1d, 0)
plt.plot(Td, y1d, label='discrete, input step 1ms');

# try with 1s sample rate

dts = 1
tds = np.arange(0.,Tc[-1]+dts,dts);
u1ds = np.zeros(tds.shape);
u1ds[0] = 1/dts

Tds, y1ds, x1ds = ctl.forced_response(sys1d, tds, u1ds, 0)
print('tds.shape',tds.shape,
      'u1ds.shape',u1ds.shape,
      'Tds.shape',Tds.shape,
      'y1ds.shape',y1ds.shape,
      'x1ds.shape',x1ds.shape)

assert td.shape[0] == y1ds.shape[0]
plt.plot(td,y1ds,label='discrete, input step 1s')
plt.legend()
plt.show()
```

repagh, 2019-02-12:
This code simply works on my computer (Fedora 28, latest slycot/python-control, python3, with some fixes for the dimensions when plotting)
Would you be able to try this on a more modern conda-based installation?
Anaconda, slycot for windows from my channel (repa), and python-control from the conda-forge channel.

p.s. one day we should get around to creating slycot for windows on conda-forge

franciscofernandezf, 2019-02-12:
> This code simply works on my computer (Fedora 28, latest slycot/python-control, python3, with some fixes for the dimensions when plotting)
> Would you be able to try this on a more modern conda-based [...]

Thank you for the answer, could you send me the links for the packages? 
I am currently using pip to install the packages, but if you think conda could work I will try it. 
I am a bit new to Python, so if you could also guide me a little through the installation it would be great.

franciscofernandezf, 2019-02-12:
> This code simply works on my computer (Fedora 28, latest slycot/python-control, python3, with some fixes for the dimensions when plotting)
> Would you be able to try this on a more modern conda-based [...]

I installed Anaconda, but when I try to install slycot from your channel I get this error:
```
(memoria) C:\Users\Francisco Fernández>conda install -c repa slycot
Collecting package metadata: done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - slycot

Current channels:

  - https://conda.anaconda.org/repa/win-32
  - https://conda.anaconda.org/repa/noarch
  - https://repo.anaconda.com/pkgs/main/win-32
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/free/win-32
  - https://repo.anaconda.com/pkgs/free/noarch
  - https://repo.anaconda.com/pkgs/r/win-32
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-32
  - https://repo.anaconda.com/pkgs/msys2/noarch
  - https://conda.anaconda.org/conda-forge/win-32
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

cgohlke, 2019-02-12:
> Wrong parameter 1 in LAPACKE_chetrs_rook

That indicates the extension was built against an older version of openblas. Recompiling fixed that. https://www.lfd.uci.edu/~gohlke/pythonlibs/#slycot

murrayrm, 2019-04-12:
Looks like this was a version issue.  Closing, but reopen if I missed something.

closed 2019-04-12 by murrayrm