python-control/python-control #288: ComplexWarning while trying to plot a pzmap

fabiomolinar, 2019-04-10:
I would like to understand why the zeros from my sys3 transfer function below aren't being plotted by the `pzmap` function.

```
#sys1
num = [6., 0., 1.]
den = [1., 3., 3., 1.]
sys = control.tf(num, den)

#sys2
n1 = [1., 1.]
n2 = [1., 2.]
d1 = [1., 2.j]
d2 = [1., -2.j]
d3 = [1., 3.]
num2 = np.convolve(n1, n2)
den2 = np.convolve(d1,np.convolve(d2, d3))
sys2 = control.tf(num2, den2)

#sys3
num3 = np.convolve(num, den2)
den3 = np.convolve(den, num2)
sys3 = control.tf(num3, den3)

control.pzmap(sys3, Plot=True, title="Mapa de polos e zeros")
```

When `control.pzmap` runs, it returns the correct poles and zeros, together with a plot. The problem is, this plot contains only the roots of the sys3 transfer function. Together with the plot, I am also getting the following warning message:

```
xxx/anaconda3/lib/python3.7/site-packages/control/xferfcn.py:864: ComplexWarning: Casting complex values to real discards the imaginary part
  num[i, j, np+1-len(numpoly):np+1] = numpoly[::-1]
```

If the problem is the removal of the imaginary parts by that part of the algorithm, why is that happening then?

fabiomolinar, 2019-04-10:
When I create my sys3 directly like this:

```
num = [6., 18., 25., 75., 4., 12.]
den = [1., 6., 14., 16., 9., 2]
sys = control.tf(num, den)
control.pzmap(sys, Plot=True, title="Mapa de polos e zeros")
```

I don't get the `warning`. But the plot still don't show the zeros of the transfer function.

murrayrm, 2019-04-11:
Which version of `python-control` are you running.  On the latest master, the plot looks right to me:
[image]
(note that there are four overlapping poles around the point -1).

I also get the warning message; not sure yet what is up with that.  It is coming from `xferfcn._common_den()` and may just be some numerical issue (that goes away when you create `sys3` directly.

fabiomolinar, 2019-04-11:
Yeah, I have calculated the poles and zeros beforehand, so I was aware about the multiple poles at -1.0.

Regarding the version, I am using 0.8.1. Looking at the repository's branches, it seems that 0.8.1 is one of the latest ones, right?

fabiomolinar, 2019-04-11:
And this is what I am seeing:

[image]

murrayrm, 2019-04-11:
Looks like this was flagged in issue #154 and fixed in PR #193.  This is scheduled for release with 0.8.2, which should be coming soon (this weekend, I hope).

murrayrm, 2019-04-11:
I chased down the source of the warning message.  The code snippet
```
d1 = [1., 2.j]
d2 = [1., -2.j]
d3 = [1., 3.]
den2 = np.convolve(d1,np.convolve(d2, d3))
```
creates a polynomial `den2` that has complex coefficients (since `d1` and `d2` are complex).  The result I get is
```
In [42]: den2
Out[42]: array([ 1.+0.j,  3.+0.j,  4.+0.j, 12.+0.j])
```
 When you then construct subsequent transfer functions, you end up with complex coefficients there as well.  For example
```
sys2 = control.tf(num2, den2)
```
gives
```
In [43]: sys2.den
Out[43]: [[array([ 1.+0.j,  3.+0.j,  4.+0.j, 12.+0.j])]]
```

Now, when you compute poles for this transfer function you get the warning:
```
In [44]: sys2.pole()
/Users/murray/Dropbox/macosx/src/python-control/murrayrm/control/xferfcn.py:896: ComplexWarning: Casting complex values to real discards the imaginary part
  num[i, j, np + 1 - len(numpoly):np + 1] = numpoly[::-1]
Out[44]: array([-3.00000000e+00+0.j,  4.57966998e-16+2.j,  4.57966998e-16-2.j])
```
If, on the other hand, you compute the poles for a polynomial with real coefficients, things are OK:
```
In [46]: control.tf([1.,3.,2.], [1., 3., 4., 12.]).pole()
Out[46]: array([-3.00000000e+00+0.j,  6.66133815e-16+2.j,  6.66133815e-16-2.j])
```

Implicitly we assume that all transfer functions have real coefficients, so the warning message is because you are creating a transfer function that violates this assumption.  We should probably generate a warning when the transfer function is created (or expand the toolbox to allow for complex transfer functions, but this is probably going overboard).