python-control/python-control #294: Implement d2d in MATLAB compatibility

tomaszG247, 2019-04-17:
I am not sure if there is an obvious way around this but would be useful to have d2d to resample a discrete time system.

Great job on the tool otherwise

murrayrm, 2021-06-10:
Great! I've tagged you on this one.  Make sure to take a look at the [python-control developers wiki](https://github.com/python-control/python-control/wiki) for more information on how to contribute.

tobin, 2024-11-26:
I came here looking for `d2c`. Has there been any progress?

KemmerN, 2025-04-08:
I would also like to see a d2c function added to the python-control. I have written a version on my own if there is any intrest I would offer looking into adding it.

bnavigator, 2025-04-08:
Please look into #888 and feel free to submit a pull request of your own idea. We will review it and then can tell if it is interesting.