python-control/python-control #303: arrows in nyquist plot

josujugo, 2019-05-22:
Recently, the Nyquist plot adds arrows to show the directions in the diagram. However, those arrows have always the same size and  the same position. Working with my students, sometimes the arrows make difficult the correct interpretation of the Nyquist plots.

New options for the arrow size and position (not always the first and the last frequencies) could help to solve the issue .

geekonloose, 2020-03-03:
Yes, I am also experiencing this problem. Is there any hack to reduce the size of arrows?
For example figure below,
[image]

murrayrm, 2020-03-03:
The problem seems to be that the arrow dimensions are based on the axis units rather than an "absolute" size (e.g., pixels).  It should be relatively easy to add an option to `nyquist_plot` to set the arrow size.  Better would be to set the arrrow size so that it isn't in axis units.  

 If someone wants to generate a PR for that, it would be useful.  I won't have time to get to it for a while.

Quick hack: you might be able to get the `arrow` object with `matplotlib`  and reset its size.

murrayrm, 2020-03-18:
PR #379 provides a quick fix that allows the arrowhead size to be set manually.

A better algorithm that computes arrow head size automatically is still needed at some point.  Arrow position is also not yet done in a good way.

SchrodingersCat00, 2020-08-19:
I'm not too familiar with the issue but the [matplotlib documentation](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.arrow.html) suggests using an [annotation](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.annotate.html#matplotlib.pyplot.annotate) with an empty string as text like this:
```python
>>> ax.annotate("", xy=(0.5, 0.5), xytext=(0, 0),
...             arrowprops=dict(arrowstyle="->"))
```
Maybe this is worth looking into?

bnavigator, 2020-08-19:
Hi @SchrodingersCat00, that sounds like a good solution, especially since the current code also just adds a dummy straight line for the `arrow` from the edges of the actual curve.

https://github.com/python-control/python-control/blob/f3e503591367f216d265fd0dd9337693923ef72d/control/freqplot.py#L534-L555

Do you want to submit a PR for that?

murrayrm, 2021-02-11:
PR #534 fixes this issue.

closed 2021-02-12 by murrayrm