python-control/python-control #306: Gain Reported by root_locus is not correct

cdboschen, 2019-05-27:
Given the following code: 
```
import control as con
gol = con.tf(1, [1, 2, 2, 0])
rlist, klist = con.root_locus(gol)
```
When I click on the root locus trajectory on the plot where it crosses the imaginary axis, the gain is reported to be approx 0.7, but I believe the gain should be 4 at this critical damping location for this transfer function. I have not reviewed the code for root_locus to find the specific issue, but wanted to document the problem if it indeed exists.

murrayrm, 2019-05-27:
Thanks for the report @cdboschen.  Which version of `python-control` are you using?  There were some updates in PR #302 that might have affected this particular bug.  If you have the ability to install `python-control` from source, perhaps checkout the latest version of master and see if the error persists?

@icam0 If you have a chance, you might want to take a look at this.

cdboschen, 2019-05-27:
I am using control 0.8.2 and python 3.7.3

I am using the Anaconda Distribution, and installed control from conda
using

> conda install control

Are you suggesting there is another variant called "python-control" that I
should try, such as

>conda install python-control

?

Thank you for all your efforts on this great tool.

murrayrm, 2019-05-27:
There is a development version of the package that  has some updated code that you could try out.  However, you would need to use git to checkout the code and then install it locally.  If you are familiar with how to do that, the version you want is [here](https://github.com/python-control/python-control) (click on  "Clone or download").  Otherwise, we'll release an updated version, hopefully with this issue fixed, as part of 0.8.3.  There is no ETA for that to come out, but probably in the next month or so since there are  a number of updates that we would like to release.

josujugo, 2019-05-27:
I have tested this issue in the same version (0.8.2) and I get a correct value.

murrayrm, 2019-05-27:
That's odd.  @josujugo, what version of Python and what version of `matplotlib` are you using?  Could be something that is not the fault of `root_locus`.

cdboschen, 2019-05-27:
This is what I did specifically (in Jupyter Notebooks) and showing what I
thought was an incorrect result.
[image: image.png]

[image: image.png]

josujugo, 2019-05-27:
@murrayrm, python version 3.7.3 and matplotlib 3.0.3

murrayrm, 2019-05-27:
@cdboschen Images didn't show up?  You may have to entire directly in GitHub as opposed to responding to the e-mail.  Can you also let us know what version of `matplotlib` you are using?

cdboschen, 2019-05-27:
Ok Will do when I get back later this evening. Thanks

murrayrm, 2019-05-30:
@cdboschen Ping...

cdboschen, 2019-05-30:
Everything is working as expected - NO ISSUE!

Hello! Sorry for my delay. I discovered the source of my issue-- it was my issue after all and not a problem with the code, so my apologies for this distraction. The actual code I used was preset with a gain of 5 as I list below, instead of the gain of 1 as I reported above.  The result in both cases (gain of 5 and gain of 1) matches expectation.

```
import control as con
gol = con.tf(5, [1, 2, 2, 0])
rlist, klist = con.root_locus(gol)
```

closed 2019-05-30 by cdboschen