python-control/python-control #308: Formatting

billtubbs, 2019-06-08:
I noticed quite a few tab characters in some python script files.

Files with tab characters:
- [external/controls.py](https://github.com/python-control/python-control/blob/master/external/controls.py) (970)
- [control/phaseplot.py](https://github.com/python-control/python-control/blob/master/control/phaseplot.py) (294)
- [control/phaseplot_test.py](https://github.com/python-control/python-control/blob/master/control/tests/phaseplot_test.py) (49)

Is there a way to globally find/replace them maybe?

Also, semi-colons are still in use or left-over from Matlab code:
- genswitch.py
- statefbk.py
- config.py
- frdata.py

I don't think they affect execution but it's not good practice to have tab characters in python scripts.

murrayrm, 2019-06-11:
It would be useful to clean these up eventually.  I and others have been updating modules to be more PEP8 compliant as we go through and implement improvements.  Fine to continue in that fashion or to selectively do PEP8 updates.  It would be good to avoid changing a lot of files at the same time, just because it makes it harder to review and comment on things.

There was an attempt to do this in PR ##283 but it was a bit too aggressive an ultimately withdrawn.

dapperfu, 2021-03-04:
One solution to fix this now, and automatically going forward is to leverage an automatic code formatting tool like `black`.

mongodb's engineering team wrote a series of posts about how they've had a lot of success with automatic code formatting tools. (They're using C++ & Clang-format, but the principles are the same). 

> But even with smaller projects, most developers discover the problems of working without an agreed upon format the very first time they work on a team. This irritation can lead to religious arguments over the merits of various formatting choices; but mature engineers know that a standard is more important than which standard. - [Succeeding With ClangFormat, Part 1: Pitfalls And Planning](https://engineering.mongodb.com/post/succeeding-with-clangformat-part-1-pitfalls-and-planning/)

I ran [`black`](https://black.readthedocs.io/en/stable/), [`reorder-python-imports`](https://github.com/asottile/reorder_python_imports), and cleaned up trailing white spaces on a branch in my fork.

This is the commit diff:

https://github.com/AutomotiveDevOps/python-control/commit/312b894a144e2e5225afbae2f72d305d845715e5

For an established project it is a big bandaid to pull off at once, but after that if you lean on the tooling then it should be smooth thereafter.

ilayn, 2021-03-04:
black is very successful in nonnumeric codea but unfortunately destroys any array alignment or longer function signatures and long string messages. So makes the code unreadable at some point which we also tried a few times on SciPy code.