python-control/python-control #309: Figure 6 in examples/pvtol-nested-ss.py is incorrect

murrayrm, 2019-06-09:
There is something wrong with the figure that is being plotted in `examples/pvtol-nested-ss.py`, Figure 6.  This should be a Bode plot of the loop transfer function with some lines for the unity gain and 180 degree phase.  But what comes out is just the gain and phase lines (no Bode plot).  This doesn't seem to be due to any recent changes, so must be something that broke a while ago.  Probably having to do with the way that "hold" and re-using axes works.

billtubbs, 2019-06-09:
Hope I didn't break it.

murrayrm, 2019-06-09:
@billtubbs I checked and it is the same before and after PR #307 and I get the same results => most have broken earlier...

murrayrm, 2020-01-03:
Fixed in PR #361

closed 2020-01-03 by murrayrm