python-control/python-control #328: The sisotool function was renamed to controlSystemDesigner in Matlab R2015a

billtubbs, 2019-07-15:
See [Mathworks release notes](https://www.mathworks.com/help/control/release-notes.html?searchHighlight=sisotool&s_tid=doc_srchtitle).

> Using sisotool to open SISO Design Tool does not generate an error in this release, but the function might be removed in a future release.

This might not be a big deal but thought I would raise it.  Assuming it was only the name that was changed shall we just add an alias for the controlSystemDesigner name?

murrayrm, 2019-07-15:
MATLAB's Control System Designer seems to be quite a bit more powerful than what is currently implemented in `sisotool`.  From the web page, it supports:

* Interactive Bode, root locus, and Nichols graphical editors for adding, modifying, and removing controller poles, zeros, and gains.
* Automated PID, LQG, or IMC tuning.
* Optimization-based tuning (requires Simulink Design Optimization™ software).
* Automated loop shaping (requires Robust Control Toolbox™ software).

All `sisotool` does at this point is let you click on the root locus plot to change the loop gain.

Having said all of that, It would be fine to just add a line at the end of `sisotool.py` to allow `controlSystemDesigner` to run `sisotool`.

billtubbs, 2019-07-15:
I see.  Well in that case if our `sisotool` is more like the 'old' MATLAB sisotool and people recognize it as that we should probably keep the name the same.  Feel free to close this issue.

billtubbs, 2019-07-15:
When I was looking at sisotool I was thinking, "this would be much better as a class object than a function".  Maybe we go down that route later and then change the name to `ControlSystemDesigner` and make it more like a fully-functional app in a window.

murrayrm, 2019-12-29:
Looks like this issue is effective resolved: leave `sisotool` since that matches the current functionality.

closed 2019-12-29 by murrayrm