python-control/python-control #332: ValueError("Time steps ``T`` must match sampling time"

bj2sn, 2019-07-23:
Got this error when using timeresp.py. Check the source code, I think the problem is at line 337-338:
if dt < sys.dt:                 
    raise ValueError("Time steps ``T`` must match sampling time")

It will be better to use the following, since float rounding error can be smaller or larger:

if abs(dt-sys.dt) < 1e-5:

murrayrm, 2019-07-23:
Can you attach a code snippet that generates the error?  This check is done in the discrete time portion of the code and so a small error between `dt` and `sys.dt` should not really occur (the only allowed differences between `dt` and `sys.dt` should be specifying a simulation sample time that is an integer multiple of the system sampling time).

Also, can you indicate which version of `python-control` you are using (`control.__version__` will tell you).  The line that you indicated doesn't seem to match the latest release.

bj2sn, 2019-07-24:
My "control" version is 0.8.2, Python version 3.7.3 (IPython 7.6.1)
The problem can be demonstrated by a simple test:

import numpy as np
samp_rate = 500e3
T = np.arange(100)
T[1]-T[0]
Out[5]: 2e-06

T[2]-T[1]
Out[6]: 2e-06

T[3]-T[2]
Out[7]: 2.0000000000000003e-06

T[4]-T[3]
Out[8]: 1.9999999999999995e-06

So, if we call timeresp.py with T[3:] as our time steps in simulation, line 265 "dt = T[1] - T[0]" will yield dt = 1.9999999999999995e-06, by sys.dt should be 2e-6. Subsequently, line 337 will raise the ValueError.

If simulation time uses T, or T[1:], or T[2:] will have no problem!

bj2sn, 2019-07-24:
It seems the problem goes away by changing line 337 in timeresp.py to:
if abs(dt - sys.dt) > 1e-18:

murrayrm, 2019-07-24:
Thanks for the extra info @bj2sn.  A fix will be a bit more complicated because the discrete time simulation routines allow you to specify a sample time for the simulation that is a multiple of the sampling time for the system.  So, for example, you can have that `dt` is `2 * sys.dt`.  On the current master, this check is done on line 362.

Will put this on my list of things to fix.

murrayrm, 2020-01-02:
Fixed in PR #356.

closed 2020-01-02 by murrayrm