python-control/python-control #333: How to use control.InputOutputSystem?

hvandonge, 2019-08-01:
How do I actually start using ```control.InputOutputSystem```?
I cannot do ```import control.InputOutputSystem```, or ```from control import InputOutputSystem```, or alike.
Can you please expand the documentation to `include` some of those basics? I am stuck at the start...

murrayrm, 2019-08-01:
A couple of comments:

* For the example above, you are doing everything with linear systems => not necessary to use the `control.iosys` module (which was intended primarily for *nonlinear* systems).  Note, however, that the current `StateSpace` class does not us named signals, so you have to use the `control.connect` function.  See issue #325 for some related discussion.

* One thing that *won't* work with the InputOutputSystems class is carrying out linear analysis operations (like `minreal`).  Instead, you would have to construct the system (using commands like the ones that you have) and the linearize it to get a StateSpace system (which you could then pass to `control.minreal`).

* An example of how to use the `control.iosys` module is in the `examples` directory, if you have the source code.  Here is a direct link:

  https://github.com/python-control/python-control/blob/master/examples/steering-gainsched.py

I'll put on my list to put the key information into the online documentation.  Hopefully the info above helps you get started.

hvandonge, 2019-08-01:
Thank you for the response.
Tomorrow I will take a better look, but at first glance, regarding your last point, I get the following result:
```
system = ct.NonlinearIOSystem()
AttributeError: module 'control' has no attribute 'NonlinearIOSystem'
```
What is causing that?

If I understand correctly, I am better off using the `connect` method in combination with indexing.
And if I really want the I/O named signals, use `iosys` and convert it to a `StateSpace`.

Again, thanks a lot!

murrayrm, 2019-08-01:
Are you using the code from the master branch (via git)?  The `iosys` module hasn't officially been released yet.

Try printing `control.__version__` and see what that says.  If you get `0.8.2`, you are using the latest release but that *doesn't* have the `iosys` module.

hvandonge, 2019-08-02:
That explains a lot, I am using 0.8.2.
In that case I will try with `control.connect`. Any idea when the new module will be released?

murrayrm, 2019-08-02:
I'll try to push something out in the next week or two.  There are a couple of things to fix in issues #329 and #332, but then it should be good to go.

Out of curiosity, where did you see the information on the `iosys` module?  If that is in any of the "default" online documentation about the package, we need to fix things up so that people don't try to access features that aren't yet avaliable.

hvandonge, 2019-08-02:
Nice, I would quite like to use it (but I am not sure how to install the non-published version, and I don't know yet if I want to ;-)).

I read about it in [https://python-control.readthedocs.io/en/latest/index.html]() under header `Library Conventions` and `Control system classes`, as I was really looking to find named connections.

One question tho, if you don't mind, about the `control.connect` method. Is it only usable for feedback blocks? I want a few signals to bypass a controller, and add them subsequently to the output of the controller, is that something one can do with `connect`?

zgh551, 2019-12-20:
My control library version is 0.8.2，but the AttributeError: module 'control' has no attribute 'NonlinearIOSystem'still hapen.i can open the iosys.py file and find the NonlinearIOSystem class.what's the question of my phenomenon

murrayrm, 2019-12-20:
The`iosys` module is not part of python-control 0.8.2.  If you want to use that module, you need to either install from source or wait version 0.8.3 is released.

Note that just downloading the source code, where you will see `iosys.py` is not the same as installing it.

zgh551, 2019-12-23:
thank for your answer ,before i used  conda install the python-control 0.8.2,so the python-control do not include the iosys file in conda.so i pull the github depository and using python install the python-control.By this way can generate the iosys file.

murrayrm, 2019-12-23:
You may have a problem with your `PYTHONPATH` and you are not getting the right version.  Try the following:
```
import control
print(control.__version__)
```
If you are importing the latest version, you should get something like
```
0.8.2.post173
```
If you are not getting the `post173` (or something like it) then you are not importing the latest version.  (The reason that the version does not list as 0.8.3 is that this has not yet been released.)

murrayrm, 2020-01-18:
The `iosys` module is now released pas part of v0.8.3 and the documentation has been updated to give more information.  Closing.

closed 2020-01-18 by murrayrm