python-control/python-control #334: LQR controller synthesis : Gain matrix K size not as expected

jsaintauret, 2019-08-05:
Hi,

first of all thank you for your hard work, this control toolbox on python has been very hepfull.

I've been using the LQR function in order to compute a K gain for my system but the size of it doesn't match the expected size.

See my system below:

A = [[-7.63943726e+03  3.20058055e-01 -2.30774129e+04  0.00000000e+00]
 [ 6.00000000e+00  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00 -1.00000000e+01  0.00000000e+00]
 [ 5.06144301e-03 -5.82851621e-05  0.00000000e+00  0.00000000e+00]]

B = [[ 9.54929659  0.        ]
 [ 0.          0.        ]
 [ 0.         10.        ]
 [ 0.          0.        ]]

C = [[ 9.99999999e-01  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  1.00000000e+00  0.00000000e+00  0.00000000e+00]
 [-2.41666102e+03  2.78291149e+01  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  1.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]]

D = [[0. 0.]
 [0. 0.]
 [0. 0.]
 [0. 0.]
 [0. 0.]]

When using the command K=lqr(A, B, Q, R) (with Q as a size 4 identity matrix and R a size 2 identity matrix), I am expecting a size 2x4 for K  since I have 2inputs in my system and 4 states variables.
here you can see the result of the command line K=lqr(A, B, Q, R)

>>> K
(array([[ 6.26241323e-04,  1.58818153e-03, -2.62861802e-03,
        -1.58677727e-03],
       [-2.75268235e-03, -1.00090821e+00,  2.83470345e+00,
         9.99998741e-01]]), array([[ 6.55798380e-05,  1.66313981e-04, -2.75268235e-04,
        -1.66166927e-04],
       [ 1.66313981e-04,  2.24611945e-01, -1.00090821e-01,
        -1.46976401e+01],
       [-2.75268235e-04, -1.00090821e-01,  2.83470345e-01,
         9.99998741e-02],
       [-1.66166927e-04, -1.46976401e+01,  9.99998741e-02,
         1.71717167e+04]]), array([-7.6393838e+03+0.j, -3.2896778e+01+0.j, -5.5096869e+00+0.j,
       -5.8285143e-05+0.j], dtype=complex64))

Any idea why I have multiple arrays? Or am I missing something?

Thank you very much

murrayrm, 2019-08-05:
The `lqr` command returns a tuple:
```
K, S, E = lqr(sys, Q, R, [N])
```
See https://python-control.readthedocs.io/en/0.8.2/generated/control.lqr.html#control.lqr

closed 2019-08-05 by murrayrm

jsaintauret, 2019-08-05:
Hi thank you for your answer and sorry for the inconvenience,
indeed I was aware of that, but since I only put K as an input, I was only expected the K part "-_-