python-control/python-control #335: Issues installing slycot via pip

atomoclast, 2019-08-08:
Hello, 

Currently trying to install `slycot` in conjunction with `control`. 

No issues installing the control library in both Python2 and Python3.

While running `sudo pip[3] install slycot`, I get the following output: 

```
Collecting slycot
  Using cached https://files.pythonhosted.org/packages/ae/9d/7ed3f2abf08aab0be9ac2b67e3040c20d9c594cce6a4af2203da0c28a6c4/slycot-0.3.5.0.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpuby90lub
       cwd: /tmp/pip-install-4y2kxn4y/slycot
  Complete output (10 lines):
  Traceback (most recent call last):
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 207, in <module>
      main()
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 197, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 48, in get_requires_for_build_wheel
      backend = _build_backend()
    File "/usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py", line 39, in _build_backend
      obj = getattr(obj, path_part)
  AttributeError: module 'setuptools.build_meta' has no attribute '__legacy__'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /usr/local/lib/python3.5/dist-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpuby90lub Check the logs for full command output.

```

I'd really like to know if this is a local issue and how to resolve it because I'm having issues using the main controls library because of errors related to needing slycot: 

```
/usr/local/lib/python3.5/dist-packages/control/statefbk.py in lqr(*args, **keywords)
    313         from slycot import sb02mt
    314     except ImportError:
--> 315         raise ControlSlycot("can't find slycot module 'sb02md' or 'sb02nt'")
    316 
    317     #

ControlSlycot: can't find slycot module 'sb02md' or 'sb02nt'

```

Thanks!

moorepants, 2019-08-08:
Maybe this helps: https://github.com/pypa/setuptools/issues/1694

atomoclast, 2019-08-08:
Thanks for the link. 

I tried looking through it, but I'm not using a venv of any sort to get it setup...
I don't have a `pyproject.toml` file in my environment as a result.

moorepants, 2019-08-08:
Ok, good to know.