python-control/python-control #337: Any plan for step_info to support negative step responses?

hgducharme, 2019-08-21:
I have step responses that go negative and sometimes stay negative, and it breaks the settling time calculation and rise time calculation. I've been messing around with it and might open a pull request if this is something y'all would be interested in. 

For example, here is my code, the step response plot, and the error

```python
from control import tf, step_response, step_info
import matplotlib.pyplot as plt
import numpy as np

transfer_function = tf( [3.32, 0, -162.8], [1, 24.56, 186.5, 457.8, 116.2] )
time, response = step_response(transfer_function)

fig, ax = plt.subplots(1, 1)
ax.plot(time, response)
plt.show()

step_info(transfer_function)
```

[image]

`step_info()` gives the following error

```
Traceback (most recent call last):
  File "bug_test.py", line 14, in <module>
    step_info(transfer_function)
  File "/Users/hgducharme/Programming/pendulum/env/lib/python3.7/site-packages/control/timeresp.py", line 536, in step_info
    SettlingTime = T[i + 1]
IndexError: index 1000 is out of bounds for axis 0 with size 1000
```

When I run this code in MATLAB, I get the following result

<hr>

[image]

murrayrm, 2019-08-21:
@joaoantoniocardoso Any interest in taking a look at this?

joaoantoniocardoso, 2019-08-21:
@murrayrm Yes, I think I can take a look at this tomorrow. 

Also I think this error is not something that happens only for negative step responses... Months ago I did a pull-request that fixes some other cases with the same message.

juanodecc, 2021-03-09:
Hi, think I solved this problem and the problem of systems with non settling time, I mean when dcgain isinf or isnan, doing:

```
    # Steady state value
    InfValue = sys1.dcgain()
    if InfValue < 0.0:
        yout = -yout
        InfValue = np.abs(InfValue)

    # SettlingTime
    sup_margin = (1. + SettlingTimeThreshold) * np.abs(InfValue)
    inf_margin = (1. - SettlingTimeThreshold) * np.abs(InfValue)
    # find Steady State looking for the first point out of specified limits
    if np.isnan(InfValue) or np.isinf(InfValue):
        SettlingTime = np.NaN
    else:
        for i in reversed(range(time.size-1)):
            if((np.abs(yout[i]) <= inf_margin) | (np.abs(yout[i]) >= sup_margin)):
                SettlingTime = T[i + 1]
                break
```

joaoantoniocardoso, 2021-03-09:
@juanodecc seems like a good solution. Could you open a PR with that, adding also a test case with a negative system and a system with non-settling time? Thanks

closed 2021-03-17 by bnavigator