python-control/python-control #338: Error", line 254, in <listcomp>     np.diff(np.diff(dstab(sys.omega[i-1:i+2])))[0] > 0 ])  IndexError: index 0 is out of bounds for axis 0 with size 0

alireza1220, 2019-09-17:
Getting this error while I am expecting to get the margins. 
I make controllers in s-domain and apply them on frd plants. most of the times it works but all of a sudden this error pops up and breaks the code.

murrayrm, 2019-09-17:
Can you include example code that generates the error?  From the error message, it looks like the error is occuring in `listcomp`, which is not a module within `python-control`

alireza1220, 2019-09-17:
I did fix it although I am not sure yet if it is a legit way of doing that. 
This error pops up when I get my widx as [ 0   37   43 ... 2493 2494 2497] . Where the first element of widx is "0".

I have fixed it by replacing 
" for i in widx if i+1 < len(sys.omega)  and i != 0 and " 

instead of yours which was
" for i in widx if i+1 < len(sys.omega)  and "

alireza1220, 2019-09-18:
I am also getting some cases where the pmidx length is zero. 
I have fixed that by adding the following. Can I do that? 
        if len(pmidx) == 0:
            pp = 0 #PM[pmidx][0] = 0    
            ww = 0 

        if len(pmidx) != 0: 
            pp = PM[pmidx][0]
            ww = wc[pmidx][0]

This at least temporary fix the issue and let the optimization not be interrupted. But if you could fix these two cases let me know to clone it again. Thanks.

closed 2019-12-23 by murrayrm