python-control/python-control #339: ImportError: cannot import name 'frange' from 'matplotlib.mlab'

DiegoPaezA, 2019-09-20:
Hi,

I'm trying to get started with the library and I always get this error. 

from control import *
import matplotlib.pyplot as plt
g=tf(1,[1,1,1])
t,y= step.response(g)
plt.plot(t,y)
plt.grid()
plt.xlab('t')
plt.ylab('t')

Thanks for your help. 
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-2-935275724fce> in <module>
----> 1 from control import *
      2 import matplotlib.pyplot as plt
      3 g=tf(1,[1,1,1])
      4 t,y= step.response(g)
      5 plt.plot(t,y)

C:\Anaconda3\lib\site-packages\control\__init__.py in <module>
     56 from .modelsimp import *
     57 from .nichols import *
---> 58 from .phaseplot import *
     59 from .pzmap import *
     60 from .rlocus import *

C:\Anaconda3\lib\site-packages\control\phaseplot.py in <module>
     40 import numpy as np
     41 import matplotlib.pyplot as mpl
---> 42 from matplotlib.mlab import frange, find
     43 from scipy.integrate import odeint
     44 from .exception import ControlNotImplemented

ImportError: cannot import name 'frange' from 'matplotlib.mlab' (C:\Anaconda3\lib\site-packages\matplotlib\mlab.py)

bnavigator, 2019-09-20:
You are using an outdated control with a recent matplotlib. `mlab.frange` was removed in matplotlib-3.1.0, and control removed the call in control-0.8.2.

Update your control package from conda-forge as described in the README and you should be fine.

DiegoPaezA, 2019-09-20:
Thanks for your help.

closed 2019-09-21 by murrayrm