python-control/python-control #340: plt.xlim won't work on limited range

limaries30, 2019-10-05:
[image]

plt.xlim doesn;t work for me on tight range.
For example plt.xlim((0,1)) works well, but when I run with plt.xlim((-0.1,0.1)) the range won't change.

Any solutions?

murrayrm, 2019-10-05:
I confirmed this behavior in the latest version of master, but not clear whether this is an issue with `python-control` or with `matplotlib`.  If someone has time, perhaps see if you can show that it works correctly if you create that plot outside of `python-control` and then we can chase down what is going wrong (and in which package)?

BTW, I tried calling `plt.xlim()` after `plot.show()` to see what happened and I got very strange results.

zachsez, 2020-05-17:
I don't have any issue, but then again I'm not sure I'm using all the right versions... How do I make sure the `import control` is the source/master fork I have instead of the released module I previously installed with pip.

murrayrm, 2020-05-17:
```
import control
print(control.__version__)
```
The master branch should say something like `0.8.3.post16`.

zachsez, 2020-05-17:
I got 0.8.1, just doubled checked and master is checkout at commit `03183d1`

murrayrm, 2020-05-17:
That's the latest commit for master, so it sounds like you are not picking up the latest version.  Try `pip uninstall control` to get rid of the old version then run `python setup.py install` in the python-control source directory to install the latest version of master.

zachsez, 2020-05-17:
Ok I think its working now its says `dev` when I print the version thank you for the help. I am able to create the problem described above but when I create my own plot for the poles and zeros its works fine...

`plt.figure(2)`
`X = [x.real for x in poles]`
`Y = [x.imag for x in poles]`
`W = [x.real for x in zeros]`
`Z = [x.imag for x in zeros]`
`plt.scatter(X, Y, s=50, marker='x', facecolors='red')`
`plt.scatter(W, Z, s=50, marker='o', facecolors='blue')`
`plt.xlim((-0.2, 0.2))`
`plt.show()`

zachsez, 2020-05-19:
So I think I figured it out, in `grid.py` in `_final_setup` `plt.axis('equal')` is called looking at the documentation [here](https://matplotlib.org/3.2.1/api/_as_gen/matplotlib.pyplot.axis.html) in order to maintain that equal aspect ratio the plot will may ignore xlim and ylim if it adjusts the figure out of an equal aspect ratio. 

Idea's:
1. We could document this in our documentation.
2. I could come up with logic similar to plt.axis('equal') which finds the max values of date and sets the limits to an equal limit to start but then allows for users to change the limits as they see fit.

murrayrm, 2020-05-20:
We should definitely do the first (at a minimum), but the second is the preferred option, I think.  There may also be a way to tell `matplotlib` to "relax" the rules for `plt.axis("equal")` that would allow rescaling of one of the axes.