python-control/python-control #342: control.lyap does not seem to be able to handle orthogonal networks

asubrama17, 2019-11-02:
It seems like `control.lyap` is not able to handle orthogonal networks, but control.dlyap can. Is there any explanation for this? Let me include a simple example where both are theoretically controllable:

N = 100
a = np.sqrt(0.9)*ortho_group.rvs(N)
b = np.eye(N)
g_disc = control.dlyap(a,b) #Works fine, gives all positive eigenvalues'
g_cont = control.lyap(a,b)

`g_cont` gives the following error:
`raise e
ValueError: The matrix A and -A have common or very close eigenvalues.`

murrayrm, 2019-11-03:
These functions use `slycot` for their computations, so not something that is directly within the `python-control` module.  In the continuous case the error is generated if `A` and `-A` have similar eigenvalues, while in the discrete case the error is generated if there are eigenvalues that are reciprocals of each other.  So this explains why you get different answers.

The reason this condition comes up in the continuous case is that in order for the Lyapunov equation to have a  solution you require that no two eigenvalues of `A` sum to zero.

Having said all of that, I ran the code above and it worked fine (importing `ortho_group` from `scipy.stats`).  So it may be that the random matrix you got was one where two eigenvalues summed to zero?

murrayrm, 2020-03-22:
No activity on this issue for a while, so closing out.  Reopen if you are still having problems.

closed 2020-03-22 by murrayrm