python-control/python-control #344: Problem with root locus using a discrete transfer function and manual delay

ezalorpro, 2019-11-09:
Hello, i am having troubles using the root locus function with a discrete transfer function and a delay, the problem occurs when the sample time is too little, i add a MWE:

```python
import control as ctrl
import numpy as np
from matplotlib import pyplot as plt

dt = 0.01 # dont work
# dt = 0.1 # it work

delay_value = 1  # delay in seconds

Gs = ctrl.tf([3.5],[10, 1])
Gs = ctrl.sample_system(Gs, dt)

delay = [0] * (int(delay_value / dt) + 1)
delay[0] = 1

system_delay = Gs * ctrl.TransferFunction([1], delay, dt)
print(system_delay)

ctrl.root_locus(system_delay)
plt.show()
```
apparently the problem is in the function _default_gains of rlocus.py, it nevers get out of the while loop:

```python
while (len(indexes_too_far) > 0) and (kvect.size < 5000):
    for counter,index in enumerate(indexes_too_far):
        index = index + counter*3
        new_gains = np.linspace(kvect[index], kvect[index + 1], 5)
        new_points = _RLFindRoots(num, den, new_gains[1:4])
        kvect = np.insert(kvect, index + 1, new_gains[1:4])
        mymat = np.insert(mymat, index + 1, new_points, axis=0)

    mymat = _RLSortRoots(mymat)
    indexes_too_far = _indexes_filt(mymat,tolerance,zoom_xlim,zoom_ylim)
```

Any idea on how to proceed?

murrayrm, 2019-11-09:
I took a quick look at this and the problem is a bit different, I think.  The loop above will always terminate since kvect.size is strictly increasing.

The underlying problem seems to be that you are giving a *very* ill-conditioned system.  I show that the transfer function as
```
      0.003498
-------------------
z^101 - 0.999 z^100
```

As a test, I changed the limit  on `kvect.size` to be 50 (so that you never enter that loop, essentially) and eventually got the following plot to show up:

[image]

ezalorpro, 2019-11-09:
oh . . ok so it is not a infinity loop, it is just a very long one due to the system characteristics, thanks for the answer, i run it on matlab and took like 15 seconds (being matlab really fast) and the plot was crazyness.

Closed

closed 2019-11-09 by ezalorpro