python-control/python-control #346: Change the input at each timestep

cmbowyer13, 2019-11-10:
Hello, I have a 2 state system I'm modeling and I want to apply a new control input at each time step, not the same input for all time steps. Can this be achieved? 

Right now it looks something like this:

# Plot response after adding feedback with optimal gain
from control import *
# Generate Responses
responses = []
states = []
timesteps = np.linspace(0,11,100)
times = []
poles = []

for t in timesteps:
    if t==0:

        response, time, state = control.forced_response(sys_ss, T=t, U=0.0)        
        responses.append(response)
        states.append(state)
        times.append(t)
        
    else:
        u = -k@state
        response, time, state = control.forced_response(sys_ss, T=t, U=u)        
        responses.append(response)
        states.append(state)
        times.append(t)
        
  The issue is it seems you have to pass in a list of times and know the control ahead of time. I want access to the state at each timestep in my simulation.

murrayrm, 2019-11-11:
You didn't include the description of the actual system you are modeling.  If all you are doing is applying linear state feedback, then the easiest thing to do is to define the closed loop system and simulate that.

You can find an example of a state feedback control in the examples/ subdirectory of the distribution.  Here is one such example:

  https://github.com/python-control/python-control/blob/master/examples/pvtol-lqr.py

cmbowyer13, 2019-11-11:
Hello, I'm modeling a single pendulum on top of a cart. So the only state variables of interest for me are the angular position from vertical and the angular velocity. I want to control it so that I can balance the pole upright. Since it is reachable, it can be reached but may oscillate about that vertical axis  from the dynamics. I want to apply linear state feedback but not use built in blocks or functions if it's possible. I'd like to apply a single input at each time step that is of the form -kx for starters and later try different inputs. The issue I'm seeing is most of the functions assume you know the input for the whole simulation time but I'd like to apply new inputs after seeing the latest system state.  I noticed two time steps are needed based on how the diff eq simulation is working under the hood.

bnavigator, 2019-11-11:
If you really want to make your own calculations between each time step, `forced_response()` does not help you much. Just calculate the next state of of your (discrete!) space state by yourself.

```python
for t in timesteps:
    x1=sys_ss.A @ x + sys_ss.B @ u
    y1=sys_ss.C @ x + sys_ss.D @ u
    u = whatever_you_want_to_achieve()
    states.append(x1)
    responses.append(y1)
    x = x1 
```

Disregarding any performance hits by doing everything in Python and extending lists of course...

BTW, to make the code in your question more readable have a look at https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

bnavigator, 2019-11-11:
But as @murrayrm noted, if you can put your feedback controller into a form that you can calculate a closed loop system, then doing a simulation with a longer time series makes much more sense. You will also have to introduce disturbances at one point.

murrayrm, 2019-11-11:
As another possibility, you can use the `InputOutputSystems` functionality to set up an arbitrary function that gets used in a simulation.  This is not part of the current release, but it is in the master branch.  Some documentation here:

  https://python-control.readthedocs.io/en/latest/conventions.html#module-control.iosys

You can find an example, including a controller block, here:

  https://github.com/python-control/python-control/blob/master/examples/cruise-control.py

cmbowyer13, 2019-11-12:
For the model problem just described, it is reachable and controllable for cts. time. 
```
import numpy as np
import matplotlib.pyplot as plt
from control.matlab import *    # Load the controls systems library
import control

A = np.array([[0, 1],
              [10.137, 0]])
b = np.array([[0],
              [-1/30]])
c = np.array([[1, 0]])

d = 0

# create the statespace model
sys_ss = ss(A, b, c, d)

# Plot step response of unstable system
(y1a, T1a, x1a) = step(sys_ss, return_x=True, X0=0)
plt.plot(T1a, y1a)
```

produces an unstable response as expected
but, when I put in my stabilized system, the
state does not remain bounded and
is not approaching the 0 upright state as designed:

```
# Create the stable system matrices
# optimal gain k precomputed:

# for desired polynomial with lambda1 = -10, lambda2 = -30
# produces this gain vector k after matching:

k = np.array([[9304.11, 1200]])
A_sta = A - b@k
print("The stabilized state transition matrix: ", A_sta)

# #Sampling Time
# ts = 0.01
sys_sta = ss(A_sta, b, c, d)

# initial condition on the state
x = np.array([[0.05],
              [0.15]])

responses = []
states = []
inputs = []

# Generate Responses
ts = 0.001
timesteps = np.arange(0,2,ts)

# Simulation with the feedback put in at every timestep
for t in timesteps:
    
    # calculate the control input
    u = -k@x
    print("u: ", u)
    x1=sys_ss.A @ x + sys_ss.B @u
    y1=sys_ss.C @ x 

    states.append(list(x1))
    print("x: ", x)
    print(x1)
    responses.append(y1)
    x = x1 

```

The response and states both explode. I am wondering why this simple example will not work? 
A 2 state system cannot be much simpler than this one.

murrayrm, 2019-11-12:
`np.linalg.eig(A_sta)` yields
```
(array([-6.83792093, 46.83792093]), array([[-0.14470407, -0.02134536],
        [ 0.98947498, -0.99977216]]))
```
So it looks like your feedback is not stabilizing (eigenvalues should have magnitude less than 1 for stability of a discrete time system).

cmbowyer13, 2019-11-12:
That is true of discrete time systems, but I've implemented the sys_ss as a continuous time state space model, so I thought as long as the eigenvalue are in the SLHP, the state should converge to the eq. pt. I will try again with forcing the eigenvalues in the unit disc and using a finer grain timestep with a discrete state space model and see if the gain works for that.

murrayrm, 2019-11-12:
If you want to implement this as an approximation of a continuous time system, you should include the time increment when you do the approximate integration, I think.

Also, the eigenvalues still have a problem since they don't have negative real part.  Perhaps something wrong in the way you are computing the gains?

cmbowyer13, 2019-11-12:
There must have been a mistake in the original gains. put after reassigning the poles to be at -1/2 for state1 and -3/4 for state2 and doing the calculation I get convergence to the origin state, and have full controllability from all initial conditions as expected. 

[image]

cmbowyer13, 2019-11-12:
I'll have a look at that nonlinear control block later and cruise control example. There's some more experiments I want to try first.

cmbowyer13, 2019-11-13:
So I think I know how to simulate any linear system now with the software, but I've moved onto to a nonlinear dynamic example (I want to model the admission control for a queue) and I have some questions about some of the blocks required and will be needed as utilized in the cruise control example. After reading the cruise control example you referenced,  I hit a few road blocks, no pun intended. 

In that example, vehicle update does the one timestep update for the nonlinear dynamics
that is f(x,u) right? And the motor_torque function is the control input for the system right?

After that point I'm not sure what the arguments to ct.NonlinearIOSystem(...) are doing and what it is used for, and why they are entered as such:
```
# Define the input/output system for the vehicle
vehicle = ct.NonlinearIOSystem(
    vehicle_update, None, name='vehicle',
    inputs = ('u', 'gear', 'theta'), outputs = ('v'), states=('v'))

# Figure 1.11: A feedback system for controlling the speed of a vehicle. In
# this example, the speed of the vehicle is measured and compared to the
# desired speed.  The controller is a PI controller represented as a transfer
# function.  In the textbook, the simulations are done for LTI systems, but
# here we simulate the full nonlinear system.

# Construct a PI controller with rolloff, as a transfer function
Kp = 0.5                        # proportional gain
Ki = 0.1                        # integral gain
control_tf = ct.tf2io(
    ct.TransferFunction([Kp, Ki], [1, 0.01*Ki/Kp]),
    name='control', inputs='u', outputs='y')

# Construct the closed loop control system
# Inputs: vref, gear, theta
# Outputs: v (vehicle velocity)
```

What are these keyword arguments doing???

```
cruise_tf = ct.InterconnectedSystem(
    (control_tf, vehicle), name='cruise',
    connections = (
        ('control.u', '-vehicle.v'),
        ('vehicle.u', 'control.y')),
    inplist = ('control.u', 'vehicle.gear', 'vehicle.theta'),
    inputs = ('vref', 'gear', 'theta'),
    outlist = ('vehicle.v', 'vehicle.u'),
    outputs = ('v', 'u'))
```

Rest is simulation/plotting but still some other blocks don't make much sense yet

```
# Define the time and input vectors
T = np.linspace(0, 25, 101)
vref = 20 * np.ones(T.shape)
gear = 4 * np.ones(T.shape)
theta0 = np.zeros(T.shape)

# Now simulate the effect of a hill at t = 5 seconds
plt.figure()
plt.suptitle('Response to change in road slope')
vel_axes = plt.subplot(2, 1, 1)
inp_axes = plt.subplot(2, 1, 2)
theta_hill = np.array([
    0 if t <= 5 else
    4./180. * pi * (t-5) if t <= 6 else
    4./180. * pi for t in T])

for m in (1200, 1600, 2000):
    # Compute the equilibrium state for the system
    X0, U0 = ct.find_eqpt(
        cruise_tf, [0, vref[0]], [vref[0], gear[0], theta0[0]], 
        iu=[1, 2], y0=[vref[0], 0], iy=[0], params={'m':m})

    t, y = ct.input_output_response(
        cruise_tf, T, [vref, gear, theta_hill], X0, params={'m':m})

    # Plot the velocity
    plt.sca(vel_axes)
    plt.plot(t, y[0])

    # Plot the input
    plt.sca(inp_axes)
    plt.plot(t, y[1])

# Add labels to the plots
plt.sca(vel_axes)
plt.ylabel('Speed [m/s]')
plt.legend(['m = 1000 kg', 'm = 2000 kg', 'm = 3000 kg'], frameon=False)

plt.sca(inp_axes)
plt.ylabel('Throttle')
plt.xlabel('Time [s]')
```

murrayrm, 2019-11-14:
The basic idea behind the `iosys` module is that you create "blocks" (`InputOutputSystem`) that have labeled inputs and outputs.  There are various ways to create an `InputOutputSystem` object (`NonlinearIOSys`, `tf2io`, etc).  Each of those objects essentially defines the right hand side of a continuous dynamical system (discrete time systems are not yet implemented).  The `InteterconnectedSystem` constructor lets you create a new `InputOutputSystem` that consists of a set of subsystems.  The keywords are describing how things are connected together.

The module needs some better documentation on how to use it, which I'll try to do when I have a chance (probably not until December), using this issue to track things (so keep posting questions and comments here).

cmbowyer13, 2019-11-14:
Ok, thanks for planning to update the documentation. And I'll keep the discussion going here after getting into the iosys module more.

murrayrm, 2019-12-23:
Closing since the original issue is resolved. Please reopen with a new title if there are problems using iosys.

closed 2019-12-23 by murrayrm