python-control/python-control #347: AttributeError: 'ValueError' object has no attribute 'info'

peterfakhry, 2019-11-14:
hi there, thanks a lot for having this powerful package. i updated recently to PYthon 3.7 and had this error upon using the control.care function that the object 'ValueError' has no attribute info. Got this error on line 517 in care, mateqn.py:
[image]

peterfakhry, 2019-11-15:
thanks a lot for your reply, but is there any corrections that we can do to get my code running?
thank you

bnavigator, 2019-11-15:
Can you post your full python Traceback?

murrayrm, 2019-11-15:
Also, note that the reason you are getting this error is that `sb02mt` is throwing an error => something is wrong in the input you are sending to that function.  You might check to make sure the matrices are the right dimensions, appropriate dimensions are full rank, etc.

peterfakhry, 2019-11-15:
thanks a lot for all, i tried to run my code again and again over the past hour but did not get that error again (dont know why :D). got this one instead
 S, eigvals, L = cl.care(A_subtelda, (Cm.T), H_subtelda, R_subtelda)

  File "C:\Users\Peter\Anaconda3\lib\site-packages\control\mateqn.py", line 503, in care
    raise ControlArgument("Q must be a symmetric matrix.")

ControlArgument: Q must be a symmetric matrix.
which is the H_subtelda matrix which i am sure that it is symmetric
[image]
thanks a lot

murrayrm, 2019-11-15:
I suspect that the issue is a precision problem.  The underlying check is
```
(asarray(Q) == asarray(Q).T).all()
```
Try computing `Q - np.transpose(Q)` and if there are nonzero entries, that is the problem.  A "fix" would be to set `Q` to be `(Q + np.transpose(Q))/2`, which should be guaranteed to be symmetric.

peterfakhry, 2019-11-15:
WOW, thanks a lot sir, actually the difference did not get a zero (as in picture). i adjusted the Q is you said and the code is running smoothly. Day saver!
Thanks a lot also' bnavigator'
[image]

closed 2019-11-15 by peterfakhry