python-control/python-control #349: coverall reports coverage decrease despite tests being added

bnavigator, 2019-11-18:
My recent pull requests enhanced some unit tests, but coverall reports a decrease in coverage.

I believe this happens because it counts the results from the travis jobs without slycot. Most of the unittests are skipped without slycot.

bnavigator, 2019-11-27:
See the reported decrease for my last contribution to #345 (commit d26d6395288d7d22181cfcfcc73baced420f52c9):

[It is compared to the master branch](https://coveralls.io/builds/2560008) : "3875 of 4625 relevant lines covered (83.78%)"
[but now counts many more lines](https://coveralls.io/builds/27285682) : "26201 of 32361 relevant lines covered (80.96%)"

Something is broken with the coveralls setup.

murrayrm, 2019-12-28:
It looks like slycot may not be getting installed properly in the runs where it is supposed to be => code that relies on slycot is no longer being tested.  Looking to see what is wrong...

murrayrm, 2019-12-30:
Fixed in PR #353.

closed 2019-12-30 by murrayrm

bnavigator, 2020-01-06:
Coverage still does not count the correct line numbers or compares to a wrong reference. See #366

bnavigator, 2020-01-28:
I don't think this is fixed by #353. With my recent PR #370, [coverall reports](https://coveralls.io/builds/28364668)

> 3952 of 4670 relevant lines covered (84.63%)

but on the [master build it is compared to](https://coveralls.io/builds/27948073)
> 34471 of 41958 relevant lines covered (82.16%)