python-control/python-control #350: Color on each branch of rootlocus

agustinvo, 2019-12-05:
When using the rootlocus function the plot containing all the branches does not differentiate between branches. All of them are colored with the same blue.

I found a walkthrough for this, but you need to know well the transfer function you want to plot, like:
`system=control.tf([1],np.convolve(np.polyadd(np.convolve([1,1],[1,1]),4),np.polyadd(np.convolve([1,3],[1,3]),4)))
control.root_locus(system)
plt.gca().get_lines()[1].set_color('b')
plt.gca().get_lines()[2].set_color('r')
plt.gca().get_lines()[3].set_color('g')
plt.gca().get_lines()[4].set_color('m')`

Where poles are the line [0], and the rest are the different branches. If you have zeros, then they are located in line[1], and the branches in [2],...

Is there any possibilty of colouring the different branches as MATLAB does?

Thank you for this module, I found it very useful!

moorepants, 2019-12-05:
I think you'd have to write a sorting routine for the computed eigenvalues because for each gain they may not be in the same order as the previous gain. For large steps between gains, I'm not sure if you can sort them, but for small gain steps you can for most cases. It isn't trivial for all bifurcation points though.

agustinvo, 2019-12-05:
I think this issue is related with this [line](https://github.com/python-control/python-control/blob/a4b4c43e51f0fc2cbf389336a90230a6a741c0dc/control/rlocus.py#L190):
'        # Now plot the loci
        for index, col in enumerate(mymat.T):
            ax.plot(real(col), imag(col), plotstr, label='rootlocus')`
where col seems to be each branch. Indeed, if `plotstr` input is modified when running the command, all the branches are plotted similarly. Its default behaviour introduces:
`# Default values for module parameters
_rlocus_defaults = {
    'rlocus.grid':True,
    'rlocus.plotstr':'b' if int(matplotlib.__version__[0]) == 1 else 'C0',
    'rlocus.PrintGain':True,
    'rlocus.Plot':True
}`

I think setting that parameter free would color each branch separatelly. I will check later on.

billtubbs, 2021-08-07:
I just tried this and it seems to work.

Replaced:
```
    'rlocus.plotstr': 'b' if int(mpl.__version__[0]) == 1 else 'C0',
```

with:

```
    'rlocus.plotstr': '' if int(mpl.__version__[0]) == 1 else '',
```

Test script:
```
system = control.tf([1], [1, 8, 30, 56, 65])
control.root_locus(system)
```

Result before:
[image]

Result after:
[image]

Would be nice to change the colour of the crosses too.

billtubbs, 2021-08-07:
I'm not sure what the benefit of the different colours is though.  Without a legend or other way of knowing what each line is, what is the benefit?

sawyerbfuller, 2021-08-07:
Colors might help track where the poles are going.

Does that coloring system still work with more complicated diagrams with
more poles and locus intersections?

Also: we might be able to deprecate matplotlib version < 1 ,  that happened
probably a decade ago by now : )

Sawyer

billtubbs, 2021-08-08:
Here's a better example which illustrates the benefit of the coloured lines.

Before:
[image]

After:
[image]

billtubbs, 2021-08-15:
One time when different colours per locus might be a bad idea, is when plotting the root loci of more than one system on the same plot.  Is this actually useful or something that we want to support?  If so, we could offer both options.

sawyerbfuller, 2021-08-15:
I like the colored branches. I had what I thought was an example where it
might fail, but I couldn’t find it.

Re multiple rloci: i think it would be good provide some sort of an
optional color-sequence argument. Which you would just set to a single
color if desired, or to your own color sequence. I think matplotlib has
features to do this, but I don’t know what it is called. S