python-control/python-control #369: Bug in xferfcn.py

rlegnain, 2020-01-27:
Hi, 
I found a bug in _common_den() method,   line 935 in xferfcn.py file. The line code is :
            num[i, j, maxindex+1-len(numpoly):maxindex+1] = numpoly

The error happen when  numerator order is larger than denominator order.  The error message is "ValueError: could not broadcast input array from shape (4) into shape (1)"
#

bnavigator, 2020-01-27:
Yes the `num` and `den` arrays are sized according to the order of the denominator

https://github.com/python-control/python-control/blob/66bee9d31c0e4913aa2bda614794845563a616d6/control/xferfcn.py#L900-L906

This *could* be extended to use the maximum of numerator and denominator, but I am not sure `TD04AD` will handle this, as it likely expects a proper transfer function? The documenation of `TD04AD` seems to imply that always `denorder > numorder`.

`_common_den()` should not fail with a `ValueError` though. A check and proper exception should be raised, or a return of the correct num and den and a warning display.

rlegnain, 2020-01-28:
I think a return of num and den with warning message will be good.  

Thanks

murrayrm, 2020-07-11:
Fixed in PR #370

closed 2020-07-11 by murrayrm