python-control/python-control #371: Complex state matrices

lytex, 2020-02-01:
Creating a system with complex state matrices is sometimes needed (for example when you are working in a diagonal basis), but unfortunately python-control is missing this functionality right now.
Since that change would likely have a lot of code modified, I suggest to create a **new branch**, **rewrite the tests with complex matrices** and see where there fail.
Maybe also it would be better to focus first on python functions and then later modify SLICOT.

bnavigator, 2020-02-01:
Most SLICOT routines expect real matrices. Support for complex state matrices in python-control would need to deal with them without Slycot.

lytex, 2020-04-01:
[Pull request ](https://github.com/python-control/python-control/pull/376)

roryyorke, 2021-09-23:
Perhaps this is well-known, but in case not: complex-coefficient transfer functions do have applications; see, e.g., [this paper by Troeng, Bernhardsson, & Rivetta, ](https://portal.research.lu.se/portal/files/30365130/2017ACC_0939_FI.pdf). The authors note that the Matlab of the time (2016) did not correctly handle complex coefficients in some cases; I guess, by implication, that it works in many other cases.

Handling this properly would be quite difficult: besides figuring out what to do with Slycot, we'd need to be sure to show responses for both negative and positive frequencies.  We'd also want to naturally combine real and complex systems without being over-eager to produce complex results.  I'm sure there are other complications I haven't thought of.

This is for the record, and for interest; I don't, alas, have time to work on it.

roryyorke, 2022-12-10:
[Root locus rules for polynomials with complex coefficients](https://ieeexplore.ieee.org/document/6608794), published 2013, and [available from one of the author's websites](https://my.eng.utah.edu/~bodson/pdf/Root%20locus%20rules%20for%20polynomials%20with%20complex%20coefficients.pdf), gives root locus rules for complex-coefficient TFs.