python-control/python-control #373: Bode Function produces wrong angular frequencies

paulvicioso, 2020-02-17:
Hi everyone,

When I try the bode function from the control.matlab library to plot the Bode of a simple function (H(s)=1/(s+1)) I'm getting the wrong angular frequencies (shifted one decade). In other words, the break frequency is at 0.1 rad/s instead of 1.0 rad/s (see image
[image]
). I'm not sure if this is an issue or I'm just doing something wrong. Could you please help me with this problem.

My system is a macOS Catalina 10.15.3 and I'm using Python 3.7.5.

```
import numpy as np
import control.matlab as ml
import matplotlib.pyplot as plt

num = np.array([1])
den = np.array([1, 1])

G = ml.tf(num, den)

print(G)

mag, phase, freq = ml.bode(G)
plt.show()
```

Thank you

murrayrm, 2020-02-17:
The units for frequency are in Hertz, not rad/sec.   1 rad/sec = 1/(2*pi) Hz = 0.16, which is where the breakpoint is (easiest to see as the point where the phase crosses 45 degrees).

It looks like this is a bug in the way that defaults are set up for the MATLAB compatibility module, which sets `bode.Hz = True`.  Can't remember if this has always been that way or whether the bug was introduced when the new `config` module was set up.

Quick fix: use `bode(..., Hz=False)`.

Proper fix: we should change the default settings to match MATLAB.

paulvicioso, 2020-02-18:
Thank you, that solved the problem.

I did not check that the frequencies were in Hz, I just made the assumption that everything was the same as in MATLAB.

closed 2020-03-25 by murrayrm