python-control/python-control #374: Bug in step_response()

rlegnain, 2020-02-20:
Hi, 
I found a bug in step_response(). An error happens when use step_response() on G(s)=1/1. Here is the code:
    u = control.TransferFunction([1], [1])  # or control.tf([1], [1])
    t, y = ctrl.step_response(u)   # error happen in this line. 

When I use scipy, it work perfectly

 The error message is:
  File "C:\Users\rlegnain\AppData\Roaming\Python\Python35\site-packages\control\timeresp.py", line 455, in step_response
    T = _default_response_times(sys.A, 1000)
  File "C:\Users\rlegnain\AppData\Roaming\Python\Python35\site-packages\scipy\signal\ltisys.py", line 2058, in _default_response_times
    vals = linalg.eigvals(A)
  File "C:\Users\rlegnain\AppData\Roaming\Python\Python35\site-packages\scipy\linalg\decomp.py", line 769, in eigvals
    homogeneous_eigvals=homogeneous_eigvals)
  File "C:\Users\rlegnain\AppData\Roaming\Python\Python35\site-packages\scipy\linalg\decomp.py", line 233, in eig
    compute_vr=compute_vr)
  File "C:\Users\rlegnain\AppData\Roaming\Python\Python35\site-packages\scipy\linalg\lapack.py", line 617, in _compute_lwork
    "%d" % (info,))
ValueError: Internal work array size computation failed: -5

bnavigator, 2020-02-20:
Can reproduce (with a different Exception raised). Properly formatted (Use the "insert code" function of github when posting code:

```
In [1]: import control

In [2]: u = control.TransferFunction([1],[1])

In [3]: u
Out[3]: 

1
-
1

In [4]: t, y = control.step_response(u)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-7a899e9160fd> in <module>
----> 1 t, y = control.step_response(u)

/usr/lib/python3.7/site-packages/control/timeresp.py in step_response(sys, T, X0, input, output, transpose, return_x, squeeze)
    514     if T is None:
    515         if isctime(sys):
--> 516             T = _default_response_times(sys.A, 100)
    517         else:
    518             # For discrete time, use integers

/usr/lib64/python3.7/site-packages/scipy/signal/ltisys.py in _default_response_times(A, n)
   2056     # TODO: This could use some more work.
   2057     # For example, what is expected when the system is unstable?
-> 2058     vals = linalg.eigvals(A)
   2059     r = min(abs(real(vals)))
   2060     if r == 0.0:

/usr/lib64/python3.7/site-packages/scipy/linalg/decomp.py in eigvals(a, b, overwrite_a, check_finite, homogeneous_eigvals)
    767     return eig(a, b=b, left=0, right=0, overwrite_a=overwrite_a,
    768                check_finite=check_finite,
--> 769                homogeneous_eigvals=homogeneous_eigvals)
    770 
    771 

/usr/lib64/python3.7/site-packages/scipy/linalg/decomp.py in eig(a, b, left, right, overwrite_a, overwrite_b, check_finite, homogeneous_eigvals)
    231     lwork = _compute_lwork(geev_lwork, a1.shape[0],
    232                            compute_vl=compute_vl,
--> 233                            compute_vr=compute_vr)
    234 
    235     if geev.typecode in 'cz':

/usr/lib64/python3.7/site-packages/scipy/linalg/lapack.py in _compute_lwork(routine, *args, **kwargs)
    804     """
    805     dtype = getattr(routine, 'dtype', None)
--> 806     ret = routine(*args, **kwargs)
    807     if ret[-1] != 0:
    808         raise ValueError("Internal work array size computation failed: "

ValueError: On entry to DGEEV parameter number 5 had an illegal value

In [5]: 
```

bnavigator, 2020-02-20:
`sys.A` is empty for a unit `TransferFunction`

ilayn, 2020-02-20:
It's on my to do list to improve `scipy.linalg.eig` API for 1.5. 0. This shouldn't have happened regardless of the control issue and the error should have been raised from eig.

I'll add an issue on the scipy side.

murrayrm, 2020-02-21:
At the same time, `step_response` probably shouldn't be computing default response times for an empty matrix (or `_default_response_times` should handle as a special case).

closed 2020-05-17 by murrayrm