python-control/python-control #381: Feature request: filter class or observer class

realteck-ky, 2020-03-18:
Thank you for a great and useful python library.
This issue is a question.

Have you planned to implement filter class or observer class?
When we implements FIR and IIR filters, LTI systems can describe their behaviors.
However, I want to obtain system response with filtering like Kalman filter under random noises.
Such adaptive filters, I think, should be implemented on `NonlinearIOSystem`.

Anyway, that is good challenge for improving.
Do you have any plan to develop filter class?

murrayrm, 2020-03-19:
A "filter" is just an input/output system.  If the filter is linear, then one of the `LTI" classes can be used (`TransferFunction` or `StateSpace`).  If the filter is nonlinear, then one of the `InputOutputSystem` classes can be used (eg, `NonlinearIOSystem`, as you suggested).

Is there a reason we would need to design a special class for filters or observers?  What would the unique operations be on this.

Closing this for now, but re-open if I am not understanding the question.

closed 2020-03-19 by murrayrm

realteck-ky, 2020-03-20:
Thanks for your reply.
I understand policy of the projects and what functions you want to provide.
I agree with your comment. My idea doesn't have enough reasons for you to be added such the class, and you don't have to re-open.

I looked forward to be released the function, `NonlinearIOSystem`, because interface of the system has good usability and versatility. I want to implement some expand class using `NonlinearIOSystem`. However, I see the layer of my challenge is different from `python-contol` providing.

Anyway, I appreciate your corresponding. Thank you.