python-control/python-control #385: Convert simple LTI state-space system from Simulink

SeanDS, 2020-03-25:
First of all, great work on this toolbox. We in the gravitational wave instrumentation community are trying to ditch MATLAB and use tools like this in Python to model control systems and dynamics. I've got what I think is a pretty basic question but I hope someone has some time to help. I hope this is the right place to post this sort of question.

I'm trying to convert a Simulink model for a triple-stage pendulum that looks like this...

[image]

...into a `python-control` system. It's a relatively simple model implemented as a state-space system (the red `SSM100g` block) with a few feedback loops and some input/output combining. I've got as far as implementing the ABCD matrices from the `SSM100g` model:

```python
import numpy as np
from control import ss

A = [...] # 36x36 matrix
B = [...] # 36x24 matrix
C = [...] # 24x36 matrix
D = np.zeros((24, 24))

SUS_TRIPLE = ss(A, B, C, D)
```

This seems to work ok, I can plot transfer functions etc. with `SUS_TRIPLE.freqresp(...)` but now I'd like to combine some of the inputs and outputs into "higher level" inputs and outputs (`act_1_x` etc.) and add some feedback (turqoise gain and sum blocks) as per the screenshot above. I see there is a ["Feedback interconnection"](https://python-control.readthedocs.io/en/0.8.3/generated/control.feedback.html) command but I don't know whether it's appropriate here. I also wonder if there is some way to abstract some of the low-level inputs/outputs of this state-space model into higher level ones, akin to Simulink's "subsystem" blocks. In the end I'd like to have another system that I can again call `.freqresp(...)` on.

Is this possible to do? Are there perhaps some examples showing how to do this? I guess this is or will be a common task in the future, converting Simulink models to `python-control`. Any help would be appreciated!

bnavigator, 2020-03-25:
Hi,

you already found the `control.feedback()` function of the block diagram algebra group. There are some more functions to define [system interconnections](https://python-control.readthedocs.io/en/0.8.3/control.html#system-interconnections) such as `control.connect()`.

Another higher level possiblity do build your model would be the new [`InterconnectedSystem` class of the `control.iosys` module](https://python-control.readthedocs.io/en/0.8.3/generated/control.iosys.InterconnectedSystem.html#control-iosys-interconnectedsystem). You put `SUS_TRIPLE` in one system and your feedback controllers into another e.g. `ECD`, where the inputs and outputs match those of `SUS_TRIPLE` and then do `SUS_TRIPLE.feedback(other=ECD, sign=-1)`.  (The docstring of [control.iosys.InterconnectedSystem.feedback](https://python-control.readthedocs.io/en/latest/generated/control.iosys.InterconnectedSystem.html#control.iosys.InterconnectedSystem.feedback) is wrong though.) You will end up with 3 additional dummy inputs for the roll, y and z feedback loops.  Make sure your signs and additions with the input signals are properly reflected in `SUS_TRIPLE` or `ECD`.

This will result in a control system object which you can call the supported functions and methods such as `freqresp()` on.

SeanDS, 2020-03-26:
Thanks a lot for your reply, @bnavigator! I'm going to attempt to use InterconnectedSystem. I need to define the ECD system though. Any suggestions how to do that? Writing ABCD matrices directly seems difficult. I guess I need to use B to map from the 24 outputs of SSM100g to the 6 internal "ECD" states, then use C to map these back to the 24 inputs of SSM100g. Is there some way to take a shortcut, e.g. with a helper function in `python-control`?

bnavigator, 2020-03-26:
You do the mapping in the constructor of `InterconnectedSystem()` by specifying tuples in the `connections=` parameter.
https://python-control.readthedocs.io/en/0.8.3/iosys.html#module-usage

Or you use `control.append()` to put `SUS_TRIPLE` and the 6  `ECD` controllers into one 30x30 system and then define `Q` for `control.connect()`

SeanDS, 2020-03-26:
Ah, I see! And how do I define the ECD system such that it behaves as a single 6x1 transfer matrix? I only see ways to instantiate systems from ABCD (state-space) or poles/zeros (transfer function).

ilayn, 2020-03-26:
Instead of -1 and ground connector and working signal-by-signal fashion, make a generic input signal Mux that collects all the reference signals (0 if direct negative feedback, leave out the columns of B if not used). Also collect all the gains into a matrix so you will have the classic feedback interconnection. Then you are back in the standard domain.

bnavigator, 2020-03-26:
> Ah, I see! And how do I define the ECD system such that it behaves as a single 6x1 transfer matrix? I only see ways to instantiate systems from ABCD (state-space) or poles/zeros (transfer function).

I guess you mean 6x6. `control.append()` works here as well. IIRC, if ECD contains just scalar gains for each signal, internally it works the same as if you collect the signals into a diagonal 6x6 (`TransferFunction` or `StateSpace.D`) matrix manually, just as Ilhan suggested

SeanDS, 2020-03-26:
Ok, I think I've implemented the system with feedback but I'm having trouble seeing any effect of the feedback.

First of all, some perhaps useful information about two aspects I found tricky while building the system: first, the system returned by `InterconnectedSystem` does not have a `freqresp` method. It seems I have to linearise it first (even though the constituent systems are themselves linear). I therefore added a call to `find_eqpt` then `linearize`. The second problem was that the system returned by `linearize` does not copy over the inputs/outputs from the system it is linearizing. This means I have to use the system I passed into `linearize` to look up the correct indices for the inputs/outputs when I plot the transfer functions. I hope that `linearize` doesn't do any remapping of the inputs or outputs.

_(I also had trouble with the docs - for example in specifying the connections, I first had all my tuples in (output, input) form but it turns out it needs (input, output) form. Maybe it could instead just detect which is an input and which is an output so order doesn't matter? Also, why does `InterconnectedSystem` take a sequence of (in, out) tuples for connections but two separate lists of `inputs` and `inplist` / `outputs` and `outlist`? Why not be consistent with `connections` and use sequences of tuples too?)_

Now the problem seems to be that the feedback isn't doing anything. I am not sure if this is because of some issue with the way I've built the feedback or because of something to do with the linearization. The responses with and without feedback are identical; I'd expect the feedback to broaden some of the resonances:

[image]

Perhaps the problem is in the way I linearise the system. I have to set parameters of `linearize` and `find_eqpt` which I just guessed and are probably wrong. Can anyone see what I'm supposed to put here? It's a linear system, and ultimately I want transfer functions from each input to each output assuming small perturbations around zero. This is really stretching my knowledge of state-space systems...

Perhaps it's useful if I post my whole script. This shows the response of an input to output in the damped and undamped case. The damped case should have broader, smaller resonances. It looks like my feedback isn't doing anything.

```python
"""100g state space model."""

import numpy as np
import matplotlib.pyplot as plt
from control import ss, linearize, find_eqpt
from control.iosys import LinearIOSystem, InterconnectedSystem

A = [
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1],
    [-152.53882610042035, 1.5571920487715416e-11, -0.011200156128132915, 1.632559549476257e-7, -1.413679342720125, -4.093359544048649e-11, 54.759209981719074, -2.003371085743382e-11, 0.01111599273426369, -2.5509597113526e-8, 2.7591338153072824, 5.412858953805167e-11, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [2.0849649640385726e-12, -145.2731502117568, 9.934442657551588e-9, 0.0003678612482892676, 9.713842032496462e-11, 0.41222315924168185, -2.6663023105712876e-12, 47.555605820245994, 4.656408908970109e-8, -0.0006654280945320643, -1.2555522934432573e-11, -0.3207695984640116, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [-0.011655967003511137, 9.83070193955856e-9, -415.4616995105061, 3.1306134749929204e-11, 0.002266377046996563, -1.422633780257092e-9, 0.0116218973819107, 4.664345630464335e-8, 147.5573202634585, -1.405363161359301e-11 [... cut ...]
# Frequency response with feedback.
mag1, phase1, _ = sys.freqresp(2 * np.pi * freq)
# Frequency response without feedback.
mag2, phase2, _ = SQL_TRIPLE.freqresp(2 * np.pi * freq)

# Plot the two cases: feedback and no feedback.
plt.loglog(freq, mag1[SQL_TRIPLE_WITH_FB.find_output("long_3"), SQL_TRIPLE_WITH_FB.find_input("act_x_1"), :], label="FB")
plt.loglog(freq, mag2[SQL_TRIPLE.find_output("long_3"), SQL_TRIPLE.find_input("act_x_1"), :], "--", label="No FB")
plt.grid(True)
plt.legend()
plt.show()
```

murrayrm, 2020-03-26:
The `InterconnectedSystem` class is supposed to maintain things as a `LinearIOSystem` if there are no nonlinear elements, but that functionality may be missing from the `InterconnectedSystem` constructor.  This would be a good enhancement and shouldn't be too hard to implement.

In the current implementation, calling `linearize` generates a `StateSpace` object, which doesn't have labeled signals.  However, it should be possible (and pretty easy) to return  a `LinearIOSystem` object, which would have the signal labels (and still be a linear system => `freqresp` would work).

Both good enhancements.  I'll try to find some time to implement, but if someone else wants to take a crack, perhaps create a separate  issue (referencing this one) and assign it to yourself?

SeanDS, 2020-03-26:
@murrayrm do you think the problem with `InterconnectedSystem` not maintaining a `LinearIOSystem` has anything to do with the lack of apparent feedback in my model?
And if I understand your reply correctly, it sounds like I shouldn't need to call `linearize` at all (if `InterconnectedSystem` maintained things as a `LinearIOSystem`). Is there some way I can hack this to work with the current `InterconnectedSystem`?

bnavigator, 2020-03-26:
> @murrayrm do you think the problem with `InterconnectedSystem` not maintaining a `LinearIOSystem` has anything to do with the lack of apparent feedback in my model?

Hi I looked into the code a little bit and it turns out that this is exactly the problem. Setting up the connection map manually does not automatically call the appropriate `StateSpace.feedback()` method for linear state space systems as iosys.feedback(ss1, ss2) would do. Right now a "feedback" modeled in an `InterconnectedSystem` would only work in a nonlinear simulation. But the `linearize()` does not use the feedback interconnection map.

> And if I understand your reply correctly, it sounds like I shouldn't need to call `linearize` at all (if `InterconnectedSystem` maintained things as a `LinearIOSystem`). Is there some way I can hack this to work with the current `InterconnectedSystem`?

Yeah we would need @murrayrm's suggestion to manintain `LinearIOSystem` by `InterconnectedSystem` and enhance LinearIOSystem to call StateSpace.feedback() with a properly reformated StateSpace system when it detects feedback loops in the connection map. Not that easy to to program in a general way, I think.

Sorry to lead you into the wrong direction. I guess @ilayn's suggestion to use a signal mux and call StateSpace.feedback() directly will be easier for now. That's basically the special case of the general approach described above.

bnavigator, 2020-03-27:
Umm, we could be completely wrong here. There is a feedback in your plot, just not very strong. I rewrote your example and just used the "classical" `control.feedback()` function for state space systems and I got the same plot:

```python
"""100g state space model."""

import numpy as np
import matplotlib.pyplot as plt
import control

A = [
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0],
    [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1],
    [-152.53882610042035, 1.5571920487715416e-11, -0.011200156128132915, 1.632559549476257e-7, -1.413679342720125, -4.093359544048649e-11, 54.759209981719074, -2.003371085743382e-11, 0.01111599273426369, -2.5509597113526e-8, 2.7591338153072824, 5.412858953805167e-11, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [2.0849649640385726e-12, -145.2731502117568, 9.934442657551588e-9, 0.0003678612482892676, 9.713842032496462e-11, 0.41222315924168185, -2.6663023105712876e-12, 47.555605820245994, 4.656408908970109e-8, -0.0006654280945320643, -1.2555522934432573e-11, -0.3207695984640116, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [-0.011655967003511137, 9.83070193955856e-9, -415.4616995105061, 3.1306134749929204e-11, 0.002266377046996563, -1.422633780257092e-9, 0.0116218973819107, 4.664345630464335e-8, 147.5573202634585, -1.405363161359301e-11, 0.0010758728848655346, -2.576128872048418e-10, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [0.00009153475730179002, 0.13768235176566324, 1.779328089502842e-8, -40.33566768070241, 4.775630208967757e-7, 0.05515222377933442, -0.000012949383725565499, -0.13768238324684695, -7.421351595388594e-9, 18.410676906038006, -1.3844040708141626e-7, -0.012669553955798114, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [-1123.897865020629, 6.31924763953394e-8, 1.816977198449706, 7.138230997697364e-7, -248.21697606442186, -2.081071479118392e-7, 605.4589919759626, -8.450027075705214e-8, -2.218607222194627, -1.2187834894017893e-7, 266.3920651963982, 2.2730013763767726e-7, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [-2.7834928034366927e-8, 144.37347739966913, -4.939652472414061e-7, 0.0073599618128027955, -8.96143076934678e-8, -125.191380162728, 5.89977725842689e-9, 78.48581645908328, 3.779265784358153e-7, -0.005915984415823647, -2.204615208168721e-9, 28.068315801949264, 0., 0., 0., 0., 0., 0., 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [119.0659799672878, -4.99222270495723e-12, 0.025241390744125194, -5.0288626114639114e-8, 1.611150201795317, 1.1477120994184478e-11, -170.41158607034222, 4.990603146133157e-12, -0.02524718274544071, 5.554015022181705e-8, -5.629379009070131, -1.3937182261466907e-11, 51.345606103054415, 8.098805932006082e-16, 5.792001384964716e-6, -5.282167744454775e-16, 0.22099954399727242, 1.3401615178387531e-17, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
    [-8.135914520461128e-9, 103.67940062530481, 1.0165117263203341e-7, 0.010094478937997822, -2.7488528895097526e-10, 0.5151551108370259, 8.135914483817255e-9, -155.02525012947387, 5.18178015683491e-7, -0.014317826247156584, 1.5831891788229847e-10, 0.4243474279077066, 3.6643873450693484e-17, 51.34584950416905, -6.198292562251438e-7, 0.005216253302724892, -6.652788379271886e-18, -0.22116445540777152, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],
[... cut ...]
           mag2[sql_outputs.index("long_3"), sql_inputs.index("act_x_1"), :],
           "--",
           label="Open Loop")
plt.grid(True)
plt.legend()
plt.show()

```

[image]
[image]

And if you increase the gain for the ecd feedback, there is more damping:

`ecd = control.ss([], [], [], 100*ecd_D)`

[image]

So it's possible that your code is completely fine and the iosys module is suitable for your task. Maybe it's the model itself. Do you expect a different output, based on your Simulink model?

SeanDS, 2020-03-27:
Ah yes, you're totally right! I wasn't looking closely enough. I was confused because MATLAB's linear analysis toolbox automatically chooses appropriate frequency vector and in this case used high enough resolution to make it obvious. Thanks a lot for your help! Much appreciated.

Do you know why what you said above, i.e.

> Setting up the connection map manually does not automatically call the appropriate StateSpace.feedback() method for linear state space systems as iosys.feedback(ss1, ss2) would do. Right now a "feedback" modeled in an InterconnectedSystem would only work in a nonlinear simulation.

is not causing a problem in this case?

What do the developers think of adding a simplified form of this code as an example to the docs? There might be others looking to translate Simulink models to Python. I am happy to clean it up a bit and make a PR if people think this would be useful.

bnavigator, 2020-03-27:
Because `linearize()` numerically simulates the dynamics of the connected model and catches the feedback. 

https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/control/iosys.py#L549-L561

https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/control/iosys.py#L1021-L1048

Depending on your system you still lose precision by converting to non-linear and performing these numerical calculations. Extending `InterconnectedSystem` to directly connect linear systems would be a good enhancement IMHO.

murrayrm, 2020-12-26:
This is a pretty old discussion (almost pre-pandemic), but there are a several enhancements suggested here and also @SeanDS volunteered to create a simplified version of this code as an example for people converting from SIMULINK to python-control, which would be great. Adding some labels to the issue so that we don't forget to look back at it (at some point...).

murrayrm, 2021-01-15:
Update: several of the features in this issue have now been implemented:

* The `linearize` function now maintains signal names.
* There is an `interconnect` function that can be used to interconnect linear or nonlinear systems using signal names.
* Interconnections of `LinearIOSystem`s via `interconnect` or via overloaded operators become `LinearICSystems` and preserve the linear structure.

With these changes, I believe that all of the new features in this thread have been implement.  The only thing missing is turning this into an example.  @SeanDS volunteered to do that (8 months ago), so I am going to close this issue and wait for a PR to come through -:).

closed 2021-01-15 by murrayrm

SeanDS, 2021-01-15:
@murrayrm Thanks for the feature implementations! I haven't forgotten about this, hopefully I can make a PR for you soon.