python-control/python-control #386: Symbolic calculation: basic Algebra error

brilliant-ember, 2020-03-30:
Say you have the following transfer function 1/( s+1 ) and you do the following algebraic operation G=a/( a+1 ) the expected output is 1/( s+2 ), but instead I get the wrong value of (s+1)/(s^2 + 3*s +2 ).
Mathmatica or any symbolic calculator proves that the output produced is wrong.
PS. I found a similar bug in Octave, I guess the implementation is referenced from one another
I believe there's an error with the symbolic evaluation.

Using control 0.8.3 and Python3 on a Linux mint machine

[image]

josujugo, 2020-03-30:
Hi, 

Using the command a/a+1, the pole/zero cancellation is not done. You will obtain the answer what you want using the command:

feedback(a,1)

murrayrm, 2020-03-30:
@josujugo is correct.  Transfer functions are not automatically "reduced"  and so if you use the overridden arithmetic operators, you can get this affect.

Closing this issue since it is resolved.

closed 2020-03-30 by murrayrm