python-control/python-control #388: Bode Plot Phase Bounded to Between -180 and 180 Degrees

Nick-Hemenway, 2020-04-03:
The output of the `bode_plot` function is bounded to between -180 degrees and 180 degrees and doesn't account for conditions when the phase should leave these ranges. For example a phase angle of -270 degrees will be represented as 90 degrees on the bode plot. To illustrate this, I've got a simple example below:

```import numpy as np
import control as ct

m = 2
k = 100_000
Wci = 2*np.pi*500

G1 = ct.TransferFunction([1],[m, 0, -k])
G2 = ct.TransferFunction([1], [1/Wci, 1])
G3 = G1*G2

ct.bode_plot([G1, G2, G3], dB = True, Hz = True)
```

[image]

-G1 (blue curve) contains two poles and a DC gain and has a phase angle of -180 degrees.

-G2 (orange curve) is a simple pole with no DC gain and thus has a phase that starts at 0 degrees and goes to -90 degrees

-G3 (green curve) is the product of the two transfer functions which on the bode plot corresponds to adding of the curves. We should see the green curve start at a phase of -180 degrees and go to -270 degrees but it does not do this because of the way that the phase angle is computed.

murrayrm, 2021-01-04:
This has been fixed in PR #494 (currently pending).

bnavigator, 2021-01-04:
It has already been merged 2 days ago.

closed 2021-01-04 by bnavigator