python-control/python-control #389: test_bdalg_functions (control.tests.iosys_test.TestIOSys) fails

flokli, 2020-04-07:
This is with:

 - python 3.7
 - python-control 0.8.3
 - matplotlib 3.1.3
 - numpy 1.18.1
 - scipy 1.4.1

```

======================================================================
FAIL: test_bdalg_functions (control.tests.iosys_test.TestIOSys)
Test block diagram functions algebra on I/O systems
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/build/control-0.8.3/control/tests/iosys_test.py", line 457, in test_bdalg_functions
    np.testing.assert_array_almost_equal(ios_y, lin_y, decimal=3)
  File "/nix/store/dh3b3w7lkcy6p56x0fbv17k0k5winx0y-python3.7-numpy-1.18.1/lib/python3.7/site-packages/numpy/testing/_private/utils.py", line 1047, in assert_array_almost_equal
    precision=decimal)
  File "/nix/store/dh3b3w7lkcy6p56x0fbv17k0k5winx0y-python3.7-numpy-1.18.1/lib/python3.7/site-packages/numpy/testing/_private/utils.py", line 846, in assert_array_compare
    raise AssertionError(msg)
AssertionError:
Arrays are not almost equal to 3 decimals

Mismatched elements: 5 / 200 (2.5%)
Max absolute difference: 0.00176677
Max relative difference: 0.33094816
 x: array([[ 0.000e+00,  1.055e-01,  2.183e-01,  3.356e-01,  4.551e-01,
         5.746e-01,  6.921e-01,  8.055e-01,  9.131e-01,  1.013e+00,
         1.105e+00,  1.187e+00,  1.257e+00,  1.316e+00,  1.362e+00,...
 y: array([[ 0.000e+00,  1.055e-01,  2.183e-01,  3.357e-01,  4.553e-01,
         5.747e-01,  6.920e-01,  8.053e-01,  9.129e-01,  1.013e+00,
         1.105e+00,  1.187e+00,  1.257e+00,  1.316e+00,  1.362e+00,...

----------------------------------------------------------------------
Ran 463 tests in 19.914s

FAILED (failures=1, errors=2, skipped=118)
```

bnavigator, 2020-04-07:
Already fixed by #366, which is merged but of course not in python-control 0.8.3.

flokli, 2020-04-07:
Thanks for the feedback! Any plans on doing a new release containing this?

flokli, 2020-04-07:
I'd also love it due to https://github.com/python-control/python-control/pull/380.

murrayrm, 2020-07-11:
Marking this with milestone 0.8.4 to keep track of the fact that it is fixed there.  Closing out as an open issue.

closed 2020-07-11 by murrayrm