python-control/python-control #396: tf2ss() returns incorrect result

chimnonvuive, 2020-05-05:
I checked the code. It seems that the problem relates to the function `td04ad` in `slycot` but I am not familiar with the language.

[image]

bnavigator, 2020-05-05:
Why do you think the result is incorrect? Both state space systems are equivalent, just a different representation.

chimnonvuive, 2020-05-06:
> Why do you think the result is incorrect? Both state space systems are equivalent, just a different representation.

Is there any

bnavigator, 2020-05-06:
If you don't have Slycot, it would be a call to `scipy.signal.tf2ss()`

https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/control/statesp.py#L984

With Slycot it is a call to `TD04AD()`
https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/control/statesp.py#L959-L960

Which is documented in the source:
https://github.com/python-control/Slycot/blob/c8d3a2143c6e0091db4801f2ee68f0d256537f2f/slycot/src/TD04AD.f#L175-L219

chimnonvuive, 2020-05-07:
> If you don't have Slycot, it would be a call to `scipy.signal.tf2ss()`
>
> https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/control/statesp.py#L984
>
> With [...]

Thank you for your generous help! I will close the issue now.

closed 2020-05-07 by chimnonvuive