python-control/python-control #398: Output matrix wrong dimensions

oayan, 2020-05-12:
Hi,

I am experimenting with the vehicle steering jupyter example. I noticed the following: 

In subsection "Linearization of lateral steering dynamics (Example 6.13)":
`# Compute the linearization at velocity 10 m/sec
lateral_linearized = ct.linearize(lateral, [0, 0], [0], params=vehicle_params)`

When I checked the output matrix of the linearized system, i.e,  `lateral_linearized.C`, it is a 2x2 matrix. This is contradicting with the book & and the explanation on this page. It should rather be a 1x2 matrix with` C = np.array([[1, 0]])`. I am wondering the source of this inconsistency.

Additionally, a minor fix:  The comment should be modified to **15/msec**, since this is the given parameter of the vehicle in the code.

Thanks

bnavigator, 2020-05-12:
I think you are confusing the `lateral_linerarized` system, which has 1 input, 2 states, 2 output, thus a `(2, 2)` output matrix `C`. With the later on printed `lateral_normalized` which only uses to first output 'y'
https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/examples/steering.ipynb#L224-L234
```python
# Define the lateral dynamics as a subset of the full vehicle steering dynamics
lateral = ct.NonlinearIOSystem(
    lambda t, x, u, params: vehicle_update(
        t, [0., x[0], x[1]], [params.get('velocity', 1), u[0]], params)[1:],
    lambda t, x, u, params: vehicle_output(
        t, [0., x[0], x[1]], [params.get('velocity', 1), u[0]], params)[1:],
    states=2, name='lateral', inputs=('phi'), outputs=('y', 'theta')
)

# Compute the linearization at velocity 10 m/sec
lateral_linearized = ct.linearize(lateral, [0, 0], [0], params=vehicle_params)
```

https://github.com/python-control/python-control/blob/6b24bb42a8da5b15cb79699e74fb33b30ba7935a/examples/steering.ipynb#L242-L245

```python
# Set the output to be the normalized state x1/b
lateral_normalized = lateral_transformed[0,:] * (1/b)
print("Linearized system dynamics:\n")
print(lateral_normalized)
```
Note the row index 0 here.

oayan, 2020-05-13:
Thank you for your comment. However:

1. First of all, according to the book, theta is not an output of the system. The system has one output, which is the lateral position 'y'. 

2. Nevertheless, if we consider it independent from the book, the output matrix 'C', which is a 2x2 matrix, is `[[1, 0], [1, 0]]`, which seems to be wrong. It could have been an eye-matrix but, having 1 and zero in both rows does not make sense to me.

bnavigator, 2020-05-13:
Okay, I see it now. Remove the 'theta' from `outputs`, which does not make sense, because `vehicle_output()[1:]` only returns 'y'.

```python
lateral = ct.NonlinearIOSystem(
    lambda t, x, u, params: vehicle_update(
        t, [0., x[0], x[1]], [params.get('velocity', 1), u[0]], params)[1:],
    lambda t, x, u, params: vehicle_output(
        t, [0., x[0], x[1]], [params.get('velocity', 1), u[0]], params)[1:],
    states=2, name='lateral', inputs=('phi'), outputs=('y')
)
```

This also obsoletes the 0 row index later in the block:
```
# Set the output to be the normalized state x1/b
lateral_normalized = lateral_transformed * (1/b)
```

bnavigator, 2020-05-13:
This means you found a bug in `control.iosys` itself. `NonlinearIOSystem` or at the latest `linearize()` should not return duplicate output rows for outputs it does not get by `outfcn`.

oayan, 2020-05-14:
Yes, thank you. I also thought that the error source might be in the lower level code than in the example itself. Should I close the issue myself or is it supposed to be done by one of the contributors?

bnavigator, 2020-05-14:
I would say leave it open until this will by fixed by changes in the notebook and `control.iosys`. I might contribute a PR later.

PR #402 (2020-05-17): Fix iosys._find_size to detect inconsistent system definitions
Fixes #398 

I also had to remove the `initial_phase=` parameter for `bode_plot()` lower down in the notebook.
closed 2020-05-17 by murrayrm