python-control/python-control #399: Bug in InterconnectedSystem

samlaf, 2020-05-14:
Hi, I found a bug in InterconnectedSystem when connecting two systems with different dt, one of which is None. I believe the correct semantic is for the new system to take the dt of the non-None system, but the way it is implemented, the order of the systems matters.

```
import control as ct
sys1 = ct.tf2io(ct.tf('s'))
sys2 = ct.NonlinearIOSystem(lambda t,x,u: x, name='sys2', states=('x'), inputs=('u'))
ct.InterconnectedSystem((sys2,sys1))
```
This code works, but changing the order of sys1 and sys2, as in `ct.InterconnectedSystem((sys1,sys2))` gives a bug:

> TypeError                                 Traceback (most recent call last)
> <ipython-input-215-d3e121778200> in <module>
>       2 sys1 = ct.tf2io(ct.tf('s'))
>       3 sys2 = ct.NonlinearIOSystem(lambda t,x,u: x, name='sys2', states=('x'), inputs=('u'))
> ----> 4 ct.InterconnectedSystem((sys1,sys2))
> 
> /usr/local/lib/python3.7/site-packages/control-0.8.3-py3.7.egg/control/iosys.py in __init__(self, syslist, connections, inplist, outlist, inputs, outputs, states, params, dt, name)
>     892                 dt = sys.dt
>     893             elif dt != sys.dt:
> --> 894                 raise TypeError("System timebases are not compatible")
>     895 
>     896             # Make sure number of inputs, outputs, states is given
> 
> TypeError: System timebases are not compatible

bnavigator, 2020-08-17:
Fixed by #431, but that will likely have to wait until 0.9.0. Reassigning milestone.

murrayrm, 2020-12-31:
I checked and #490 (rebased #431) fixes the initial bug that was reported regarding mixed timebases.

However, it uncovered another bug: the command `sys1 = ct.tf2io(ct.tf('s'))` tries to generate a state space system having transfer function s (differentiator).  This is not possible since you state space systems always have proper transfer functions and s is not proper => it should generate an error.

Leaving this bug open until this is fixed.  I'll also update the title.

closed 2020-12-31 by bnavigator