python-control/python-control #407: stability_margins Fails for Many FRD Systems at Stab Margins

stevecohen42, 2020-05-25:
There is a fairly obvious copy-and-paste error in the stab margins section for FRD systems in stability_margins. The calculation for the wstab array checks the indexes only for `i+1 < len(sys.omega)`, copied from the phase margin test, and then goes on to access `sys.omega[i-1:i+2]`. Clearly this should be checking both that `i+2 < len(sys.omega)` and that `i-1 >= 0`.

So to correct this, line 244 in margins.py becomes something like:
```
              for i in widx if i+2 < len(sys.omega) and i-1 >= 0 and
```

bnavigator, 2020-05-26:
Here is the relevant code:

https://github.com/python-control/python-control/blob/03183d157a5f6cd8a9918498ab8717dc7b91c9cf/control/margins.py#L222-L224

https://github.com/python-control/python-control/blob/03183d157a5f6cd8a9918498ab8717dc7b91c9cf/control/margins.py#L240-L247

In Line 241, `np.diff` is called twice, reducing the length of the input to `np.where` by 2. So the maximum possible `i` in `widx` is already smaller than `len(sys.omega)-2`.

You are correct though for the `i==0` case. That fails in line 245 because `sys.omega[-1:2]` returns an empty array and indexing is not possible then.

stevecohen42, 2020-06-29:
Right - so the i+1 case in the current code is superfluous and the i-1 case needs to be added.

PR #425 (2020-07-11): Fix margin indexing
Fixes #407
commit 711a5128 (2020-07-11): Fix margin indexing bug in FRD stability margin computation + code cleanup (#425)

* fix #407 (stability_margins Fails for Many FRD Systems at Stab Margins)
* clean up indexing
* add margin index test
* reorganize and clean up margin_test.py
closed 2020-07-11 by murrayrm