python-control/python-control #408: Enhancement Proposal: autodiff linearization in IOSystems

samlaf, 2020-05-25:
I noticed that the `linearize` function in iosys.py uses finite difference.
I think it wouldn't be too hard to implement an autodiff-based linearize, using autograd, and I propose to do that. I would change the interface to something like
```def linearize(self, x0, u0, t=0, params={}, eps=1e-6, method='finite diff'):```
and changing `method='autograd'` would use autodiff instead (keeping finite diff as the default so that users don't need to install autograd if they don't want to).

Is there any reason why autodiff wasn't used? Maybe numerical reasons like finite differences do better in the low dimensional types of systems that we see in control?
If this change seems reasonable and valuable, I am requesting permission to make this change.

moorepants, 2020-05-25:
> Is there any reason why autodiff wasn't used?

The growth of Python based autodiff tools is more recent than the origin of Python Control. You would have essentially had to roll your own when many of the core scientific python libraries were written.

Having an optional dependency on libraries like autograd or jax may be a good idea, but those libraries are also relatively young relative to the other dependencies: numpy, scipy, etc., so some care should be taken when setting up dependencies on them.

moorepants, 2020-05-25:
Note that autograd does not seem to be maintained anymore: https://github.com/HIPS/autograd/graphs/contributors?from=2019-11-05&to=2020-05-25&type=c

samlaf, 2020-05-25:
Ah I didn't know. Just realized it's mentioned on their github page:

> Note: Autograd is still being maintained but is no longer actively developed. The main developers (Dougal Maclaurin, David Duvenaud, Matt Johnson, and Jamie Townsend) are now working on JAX, with Dougal and Matt working on it full-time. JAX combines a new version of Autograd with extra features such as jit compilation.

samlaf, 2020-05-25:
What did you have in mind when you mentioned "some care should be taken when setting up dependencies on them"?

I was thinking of doing something like pymanopt have done, where they set up a generic "backend" dependence, and then function calls can specify which backend they want to use.

For example, https://github.com/pymanopt/pymanopt/blob/master/examples/dominant_eigenvector.py.

```
def create_cost_egrad(backend, A):
    m, n = A.shape
    egrad = None

    if backend == "Autograd":
        @pymanopt.function.Autograd
        def cost(x):
            return -np.inner(x, A @ x)
    elif backend == "Callable":
        @pymanopt.function.Callable
        def cost(x):
            return -np.inner(x, A @ x)

        @pymanopt.function.Callable
        def egrad(x):
            return -2 * A @ x
    elif backend == "PyTorch":
        A_ = torch.from_numpy(A)

        @pymanopt.function.PyTorch
        def cost(x):
            return -torch.matmul(x, torch.matmul(A_, x))
    elif backend == "TensorFlow":
        x = tf.Variable(tf.zeros(n, dtype=np.float64), name="X")

        @pymanopt.function.TensorFlow(x)
        def cost(x):
            return -tf.tensordot(x, tf.tensordot(A, x, axes=1), axes=1)
    elif backend == "Theano":
        x = T.vector()

        @pymanopt.function.Theano(x)
        def cost(x):
            return -x.T.dot(T.dot(A, x))
    else:
        raise ValueError("Unsupported backend '{:s}'".format(backend))

    return cost, egrad
```

samlaf, 2020-05-25:
Also, out of curiosity, is there a problem with having dependences on non actively-maintained repos? From my (very limited) experience autograd has always been the easy to install and use "small-brother" autodiff package, as opposed to other monsters like pytorch, tensorflow or jax.

moorepants, 2020-05-25:
> What did you have in mind when you mentioned "some care should be taken when setting up dependencies on them"?

I meant to imply that we shouldn't add required or optional dependencies unless the python-control developers are ready to maintain that support.

> Also, out of curiosity, is there a problem with having dependences on non actively-maintained repos? 

Yes, there is a problem. As python-control moves forward if the dependencies don't move forward it then becomes a maintenance burden to the python-control-developers. For example, eventually autograd will not work with the current versions of Python, NumPy, SciPy, etc. that python-control supports. The term "technical debt" describes this: https://en.wikipedia.org/wiki/Technical_debt

tmmsartor, 2020-05-25:
Given the fact you are looking into automatic differentiation, I would suggest to look into [CasADi](https://github.com/casadi/casadi) as well. It implements fast automatic differentiation and it provides interfaces to non-linear optimization solvers. It can be used also to solve Optical Control Problems.
It's is general more oriented toward dynamical systems than big neural networks.