python-control/python-control #429: iosys failure with numpy 1.19 due to changed API

bnavigator, 2020-07-09:
Numpy 1.19 changed indexing for `np.delete()`, so `iosys.find_eqpt()` started to fail:

```
[  117s] =================================== FAILURES ===================================
[  117s] __________________________ TestIOSys.test_find_eqpts ___________________________
...
[  117s] >       xeq, ueq = ios.find_eqpt(
[  117s]             nlsys, [0, 0, 0, 0], [0.01, 4*9.8], y0=[0.1, 0.1], iu = [])
[  117s] 
[  117s] control/tests/iosys_test.py:607: 
[  117s] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[  117s] control/iosys.py:1647: in find_eqpt
[  117s]     input_vars = np.delete(np.array(range(ninputs)), iu)
[  117s] <__array_function__ internals>:5: in delete
[  117s]     ???
[  117s] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[  117s] 
[  117s] arr = array([0, 1]), obj = array([], dtype=float64), axis = 0
[  117s] 
[  117s]     @array_function_dispatch(_delete_dispatcher)
[  117s]     def delete(arr, obj, axis=None):
[  117s]         """
[  117s]         Return a new array with sub-arrays along an axis deleted. For a one
[  117s]         dimensional array, this returns those entries not returned by
[  117s]         `arr[obj]`.
[  117s]     
[  117s]         Parameters
[  117s]         ----------
[  117s]         arr : array_like
[  117s]             Input array.
[  117s]         obj : slice, int or array of ints
[  117s]             Indicate indices of sub-arrays to remove along the specified axis.
[  117s]     
[  117s]             .. versionchanged:: 1.19.0
[  117s]                 Boolean indices are now treated as a mask of elements to remove,
[  117s]                 rather than being cast to the integers 0 and 1.
...
[  117s]                 keep = ones(N, dtype=bool)
[  117s] >               keep[obj,] = False
[  117s] E               IndexError: arrays used as indices must be of integer (or boolean) type
[  117s] 
[  117s] /usr/lib64/python3.8/site-packages/numpy/lib/function_base.py:4406: IndexError
```

murrayrm, 2020-07-09:
Thanks, @bnavigator.  Looks like we just need to change the argument to a list with one element and should work with old and new numpy.

bnavigator, 2020-07-09:
I am preparing a PR

PR #430 (2020-07-11): only call np.delete with actual removal
Fixes #429

`ix` and `iu` can be empty lists or numpy arrays, which trips the new `np.delete`.
closed 2020-07-11 by murrayrm