python-control/python-control #451: Release 0.8.4 and 0.9.0 sequencing

murrayrm, 2020-08-16:
I'm opening this issue to create a place to talk through how we are going to merge the many PRs that have built up over the past month (which is great!) in an order that makes sense (which might be hard).

As a starter, it would be good to tag PRs as something that can go in release 0.8.4 (patch level change => no existing code should stop running) versus 0.9.0 (minor version change => some code might need to be adjusted to work with the release).

I'll update this post later to try to include a list of the existing PRs, any dependencies, and which of these two releases they should go with.

murrayrm, 2020-08-16:
Tried adding @bnavigator to this issue, but not showing up on the list.  Mentioning here instead.

bnavigator, 2020-08-16:
@sawyerbfuller  goes by a different handle now?

bnavigator, 2020-08-16:
#438 includes a merge strategy and resolves conflicts for the PR's it depends on, if you want to take that as a guide. The PR itself is thus only suited for >=0.9.0

sawyerbfuller, 2020-08-17:
hi Richard, i lost control of the blinkminster GitHub account when I lost my caltech email address. Any idea how I can mark pr's with a milestone?

murrayrm, 2020-08-17:
I can add your new account as a developer (and remove the old one).

murrayrm, 2020-08-17:
OK, you should now have developer privileges @sawyerbfuller.  Let me know you can't set milestones.

sawyerbfuller, 2020-08-19:
I'm happy with milestones as they're currently set. 

From my perspective, once 0.8.4 is out, can we merge #438 and its dependencies into the dev branch? With that I can take care of merging #449, which makes changes in a lot of places and still does the old style unit testing. #454 might need similar treatment from me, but won't be as involved.

bnavigator, 2020-08-19:
#438 has merge conflicts now, because the PRs tagged for 0.8.4, which have already been merged into master, have a different sequence than the current merge order in #438. I will need to rebase #438.

### Proposal
Declare (a rebased) #438 to be the first merge after 0.8.4 and have all other PRs, specifically #400, #431 rebase onto that. 

We could also open a real "develop" or "0.9.x" branch even before releasing a 0.8.4 tag on master. The open PRs for 0.9 could then be based against that branch, where a reviewed #438 is merged first and the other PRs are merged after review too. No need to block future development until 0.8.4 is released.

sawyerbfuller, 2020-08-19:
Having a develop branch seems reasonable to me (and I think is very common? no personal experience), but I'm also happy to wait to merge until after 0.8.4 is out.

murrayrm, 2020-08-20:
I've been hesitant in the past to create version specific branches, just because then we have to merge PRs onto multiple branches (eg, if someone submits a PR that is relevant to 0.8.4 we have to merge into master than then remember to merge into develop/0.9.0. 

But maybe we are at the point where having a single master is slowing us down?

sawyerbfuller, 2020-08-20:
Happy to give it a try. Might be the case that the time period during which there are multiple branches is pretty short, and would collapse back to a single master after bug fix releases are out.

But I'm not sure. Git is pretty complicated and there are already enough obstacles for the average controls engineer -- who isn't likely to be a git guru -- to contribute. I think having a single master branch makes that a little bit simpler.

bnavigator, 2020-08-20:
After you merge something into master you can always merge master into develop.

Some projects do it the other way round. They have one or more release branches which get merges from default master/main/develop. This means that the release tags are not reachable from master.

We don't have to do this. Version 0.8.4 is around the corner, right?

sawyerbfuller, 2020-10-07:
I just realized I had started a review of #438 awhile back but hadn’t submitted it, so I just did. Looks like there are a few problems in there (maybe from merging with an old version of a PR ) that might take a little work to fix. I think it might be best to leave that out of 0.8.4, do a release sooner rather than later, and then work in the dev branch to work out the changes needed in #438.

murrayrm, 2020-10-15:
Spending some time trying to catch up on the current state of this discussion.

First, some data.  Here's a summary of the current PR's that look like they should go into 0.8.4:

* #400, I/O systems naming bugs: can merge into 0.8.4; no major conflicts with anything
* #435 legacy default functionality
* #454 improved default time vector for time response functions that takes zeros into account

I have done a local merge of these 3 PRs are there are no conflicts and all tests past.  Any objections to merging these (one at a time, confirming CI passes at each stage), releasing 0.8.4, then moving on to the larger list of 0.9.0 PRs (starting with #438, which updates the test suite)?

bnavigator, 2020-10-15:
I will need to update #438 after those three PRs have been merged, but that's okay. My change request in #400 will bee moot and I can do it myself.

sawyerbfuller, 2020-10-15:
I tried to incorporate a suggestion to the unit test suggested by @bnavigator back in August into #454 this morning, but having git troubles and it's not that important. So I'd say your plan as written sounds great to me.

murrayrm, 2020-10-17:
I've merged the three PRs listed above.  

There are still a few things left to do before we can release [0.8.4](https://github.com/python-control/python-control/milestone/7):

* Go through open issues and tag anything that needs to be fixed now (and won't generate incompatibilities) as [milestone 0.8.4](https://github.com/python-control/python-control/milestone/7)

* Check and fix a bug in the `bdalg.connect` function (issue #421).  This looks like a matter of adding some error checking for indices that are out of bounds.

* Add some warning messages to `matlab.markov` as described in issue #395.  The real fix here is something that will be part of 0.9.0.

* Documentation updates: I have some collected documentation updates that I need to rebase and apply.  When I do that I'll take care of open issue #418 (doc inconsistency in return type).

Other than that (and assuming the first item doesn't generate a whole slew of issues), we should be about done.

sawyerbfuller, 2020-10-20:
SOunds good to me. Hopefully #421 is not too bad. I've been procrastinating looking at it for fear that it may entail fixing #244. But maybe that won't be necessary, and it would be great to have a fix.

sawyerbfuller, 2020-12-19:
OK I think #421 is fixed in #474 . Wasn't too bad.

bnavigator, 2020-12-21:
> There are still a few things left to do before we can release [0.8.4](https://github.com/python-control/python-control/milestone/7):
> 
> - [x]  Go through open issues and tag anything that needs to be fixed now (and won't generate incompatibilities) as [milestone 0.8.4](https://github.com/python-control/python-control/milestone/7)
> - [x] Check and fix a bug in the `bdalg.connect` function (issue #421).  This looks like a matter of adding some error checking for indices that are out of bounds.
> - [ ] Add some warning messages to `matlab.markov` as described in issue #395.  The real fix here is something that will be part of 0.9.0.
> - [ ] Documentation updates: I have some collected documentation updates that I need to rebase and apply.  When I do that I'll take care of open issue #418 (doc inconsistency in return type).

murrayrm, 2020-12-26:
I went through the remaining issues and I think the only thing that we need to fix before releasing 0.8.4 is issue #395, which is addressed in PR #478.  As soon as that is merged, we can go ahead with the release.  Hoping to do that in the coming week.

murrayrm, 2020-12-28:
Version 0.8.4 is now released and available on PyPI and conda-forge.  I suggest the following moving forward:

* We create a branch 0.8.x just in case there are thing that need to be fixed in 0.8.4 before we are ready to release 0.9.0

* We update the master branch to be 0.9.x, so that developers who check out the master branch build on the latest code.

I think we can leave this issue open until we have a "stable" master that implements the changes here.

murrayrm, 2020-12-30:
@bnavigator @sawyerbfuller I'll have some time today to review pull requests, if that's useful.  Is there an order that we are going to use for merging these in to minimize rebasing work and who should pull the trigger on the PR merges?

bnavigator, 2020-12-30:
#438 is rebased and ready but needs #485 to pass the tests. #486 needs #438.

As soon as a PR is reviewed (by comments or Github tag) and passes tests, I would say anyone can merge, agreed?

Proposed order:

- [ ] #485
- [ ] #438
- [ ] #486

These need to be rebased. Better do it after #438:

- [ ] #431
- [ ] #449 
- [ ] #450

- [ ] move #475 to the wiki 
- [ ] extend #484 with (new pytest style) unit tests. Supersedes #376

murrayrm, 2020-12-30:
@bnavigator I will merge #438 as soon as the CI on the PR completes, since it is holding everything else up.  I may trim out some of the Python version tests quickly after that, so cut down CI delays.

bnavigator, 2020-12-30:
Okay, I was in a meeting the last 3 hours.

You squashed the commits, so I will need to keep my branch and revert 2b98769c9e51ca45291b39c6e5fa53f27a6357d9 for #431. Since @sawyerbfuller is not able to contribute code currently, I will look at #431 and take up the task of rebasing it.

murrayrm, 2021-01-14:
Once we close out the remaining 0.9.0 issues and PRs, I think we can close off this thread.  Some open issues/questions:

* I'm not sure when we want to release 0.9.0.  It might break a lot of code, so having it be the default version only a few weeks after 0.8.4 came out might not be wise?  Unfortunately, I don't see a way to upload 0.9.0 to conda-forge and not have it be the default version.  Any thoughts on timing?
* Are there any other "big" changes that we want to put into 0.9.0 before releasing it? Anything that is pending and that would break user code seems like a good thing to put in early so that we can do "patch" updates (0.9.1, 0.9.2) for a while after that.
* Related to issue #446 (and PR #504), I was planning to shut down Travis CI this weekend and just rely on GitHub Actions after that.  We can leave the `.travis.yml` file in place in case we need to go back.  The main thing we lose is Python 2.7 testing (so we will break something against 2.7 quickly, I would guess).

bnavigator, 2021-01-14:
> * I'm not sure when we want to release 0.9.0.  It might break a lot of code, so having it be the default version only a few weeks after 0.8.4 came out might not be wise?  Unfortunately, I don't see a way to upload 0.9.0 to conda-forge and not have it be the default version.  Any thoughts on timing?

GitHub and PyPI allow "pre-releases". Is that a feature for conda too? My hate for conda increased during the Github Actions efforts, and I do not use it on my Linux box, so I don't want to get involved into it too much.

> * Are there any other "big" changes that we want to put into 0.9.0 before releasing it? Anything that is pending and that would break user code seems like a good thing to put in early so that we can do "patch" updates (0.9.1, 0.9.2) for a while after that.

What about the complex state space matrices (#484)? Users have been demanding it. (#371, #468)

> * Related to issue #446 (and PR #504), I was planning to shut down Travis CI this weekend and just rely on GitHub Actions after that.  We can leave the `.travis.yml` file in place in case we need to [...]

Given that almost all recent PRs had to introduce some kind of workaround for Python 2 in the process, I am all for it.

sawyerbfuller, 2021-01-14:
0.9 as it is currently has most of the big changes I had in mind. There are a few other things on the to-do list, but they will be fine for minor releases:

* discrete-time margin has problems with large negative margins
* margin may not work with ss systems
* signal names in in TF and SS systems
* dlqr, dlqe

Farther off (and may not break code):
* time delays

Fine with me to stop testing against 2.7.

murrayrm, 2021-01-14:
Good list.  I don't think any of those will break existing cost => probably OK for patch-level changes.

For signal names: that is already in place via the `LinearIOSystem` class and the `interconnect` function. Do we need something substantially different?

sawyerbfuller, 2021-01-14:
re timing: 0.9 prerelaes sounds interesting. but if its going to break code its going to break code now or six months from now. 

re signal names: possible to turn that lineario system into a TF or SS system? assuming so, it seems a bit roundabout, but definitely a nice feature

murrayrm, 2021-01-14:
All LinearIOSystems are also StateSpace systems => everything can be done and you get signal names.

sawyerbfuller, 2021-01-15:
Oh yeah, forgot about #244 ! 

A few more non-breaking:
* system to simulate interconnection of discrete and continousus time systems ala simulink, and possibly create a figure showing block diagram using matolotlib. probably enough to use `interconnect`. can a discrete-time system preceded by A2D and followed by D2A be modeled using a nonlinearIO sytem? TBD
* rhs and out methods for numerical simulation of ss systems 
* fill in missing elements in @roryyorke’s table in #459

bnavigator, 2021-01-15:
Should we consider setting up milestones and [projects](https://github.com/python-control/python-control/projects) for these?

murrayrm, 2021-01-18:
Quick updates:

* Travis CI is now turned off (removed the webhook)
* Resolution or issues for some of the items pointed out by @sawyerbfuller are in place:
  - signal names in in TF and SS systems: use `LinearIOSystem` class (see above)
  - dlqr, dlqe: issue #359 is now tagged as 0.9.x
  - fill in missing elements in @roryyorke’s table in #459: created a new issue (#513) to track and assigned to milestone 0.9.x

It would be good to get issues set up to track the other suggestions above so that we don't lose track of them.

sawyerbfuller, 2021-01-19:
one more possible breaking change idea: sys.inputs and sys.outputs are different things in ss and iosys: numbers-of vs names-of, respectively. i think most would agree iosys convention, using ninputs/noutputs, is proabably a better attribute name. this is baked into a lot of places in the library though, and in user code, but easy enough to do a search and replace. 

may not be worth the friction to make a change to iosys convention everywhere, but since we are talking abiut it, i thoight i would bring it up for discussion. What do you think? worth the trouble?

murrayrm, 2021-01-19:
I agree that having the names be different is confusing and I like `ninputs`, `noutputs`, and `nstates` over the `LTI` equivalents. It could break people's code and so if we are going to change it, this might be a good time to do so.

bnavigator, 2021-01-19:
Sorry I have been busy with other things and didn't keep up with the squeeze discussions. Just a quick idea regarding `inputs`. I also like the idea to switch over to `.ninputs` and co. We can replace it in our own code pretty quick and could introduce deprecation warnings for legacy `.input` usage by making them class properties with getter and setter functions. Until we replace them to contain signal names in a future version.

murrayrm, 2021-01-20:
`inputs` to `ninputs` in PR #515.

murrayrm, 2021-01-20:
OK, looks like everything in the discussion above that matters for [0.9.0](https://github.com/python-control/python-control/milestone/8) are addressed except for whether we want to include complex state space matrices or not (#484). It looks like that one is going to take considerable effort, so maybe we close out issue, merge pending requests #514 and #515 (after review and updates), tag 0.9.0 (but perhaps not release it, so that it is still "internal"?), and then start working on 0.9.1.

Thoughts?

sawyerbfuller, 2021-01-22:
I dont know what the use case is for complex state space matrices is, so no biggie for me if that gets pushed off while more substantial revisions are done/contemplated ot make that possible. 

I have been hoping to that the library will be functional enough to be offered as an option for students in a digital (discrete-time) control class (UW AEM581) I am offering in the spring quarter. The good news is that in the current dev branch, things are now mostly functional for that purpose!  

Having a 0.9.x public release some time this quarter would be nice so that install instructions are simple and "future-proof". But no biggie, it would be easy enough to have them download the dev branch.

murrayrm, 2021-01-23:
Look like instruction begins 29 Mar 2021, right?  If so, we can definitely make 0.9.0 (or later) available by then.

murrayrm, 2021-02-25:
I put some release notes for 0.9.0 on the [release notes](https://github.com/python-control/python-control/wiki/Release-notes) page on the wiki.  As soon as we merge #549 and possibly #546, we can release 0.9.0.

murrayrm, 2021-03-13:
Update: as soon as we merge PR #558, #566, and (optionally) #569, I think we are good to go.  I'll target doing the release starting around 21 Mar.

sawyerbfuller, 2021-03-14:
Sounds great. S

murrayrm, 2021-03-20:
All changes are in.  Release notes are updated.  I'll start the (somewhat slow) process of release 0.9.0 tomorrow (20 Mar) and then close out this issue when done.

sawyerbfuller, 2021-03-20:
Excellent

murrayrm, 2021-03-20:
python-control 0.9.0 now released.

closed 2021-03-20 by murrayrm