python-control/python-control #460: Bug in forced_response() function

rlegnain, 2020-08-27:
Hi, 
I have found a bug in forced_response() function when pass time vector to it. 
Here is the code:
`
import control
import numpy as np
import matplotlib.pyplot as plt

if __name__ == '__main__':
    t = np.linspace(0, 0.03, 0.01)
    H = control.TransferFunction([1], [1, 100])
    t, y = control.step_response(H, T=t)
    plt.plot(t, y)
    plt.grid()
    plt.show()
`

I got the following error:
Traceback (most recent call last):
  File "C:/w/siad/backend/mytest_new_lib.py", line 8, in <module>
    t, y = control.step_response(H, T=t)
  File "C:\Python35\lib\site-packages\control\timeresp.py", line 534, in step_response
    squeeze=squeeze)
  File "C:\Python35\lib\site-packages\control\timeresp.py", line 311, in forced_response
    dt = T[1] - T[0]
IndexError: index 1 is out of bounds for axis 0 with size 0

sawyerbfuller, 2020-08-27:
Hi Rajab, I am not positive, but I am pretty sure your time vector t is
empty the way you have defined it. Sawyer

Sawyer Fuller
Assistant Professor of Mechanical Engineering
Adjunct, Paul G Allen School of Computer Science
University of Washington
http://faculty.washington.edu/~minster/

(Typed with my thumbs)

rlegnain, 2020-08-27:
@sawyerbfuller 
Sorry,  my mistake :(
Many thanks

closed 2020-08-27 by rlegnain