python-control/python-control #461: control.iosys.NonlinearIOSystem generates "Algebraic loop"

Arch111, 2020-10-01:
Hi,

Issue Definition:

1) Define a nonlinear system with an output function and NO update function
2) In the output function, define the output as a random number (or any random output)
3) After running the simulation You get "RuntimeError: Algebraic loop detected.". 

I guess you have to be able to define any deterministic system as long as you can define it properly 
as a black box which maps an input to an output, during the simulation

How to generate the issue:

1) Take the example code at https://python-control.readthedocs.io/en/0.8.3/steering-gainsched.html
2) At line 113, modify the statement "return np.array([vref * t, yref, 0, vref, 0])" to "return np.array([vref * t, yref, 0, vref*np.random.rand(), 0])"
3) Run the code

I get the same error whenever I define my custom black box function. If this is an expected behavior for control.iosys.NonlinearIOSystem, is there a way to define and insert a black box system to the control.iosys.InterconnectedSystem so that it can cooperate with other NonlinearIOSystem type systems?

Thanks

murrayrm, 2020-12-26:
The I/O system functionality only works for deterministic systems and the behavior you are illustrating is non-deterministic.

The specific problem with the example that you provide is that every call to the `trajgen` block (which is the one that you modified) generates a different random number.  So when the simulator runs and tries to make sure that any feedthrough terms have been properly processed, the algorithm can't conference.

Here's the comment in the code describing what the algorithm does (from `iosys.py`, line 1062):
```
# To allow for feedthrough terms, iterate multiple times to allow
# feedthrough elements to propagate.  For n systems, we could need to
# cycle through n+1 times before reaching steady state
# TODO (later): see if there is a more efficient way to compute
```

I don't think this is an issue that can be addressed in the current way that the I/O simulation runs.  I'm closing this issue, but re-open if I have not interpreted what you are trying to do correctly.

closed 2020-12-26 by murrayrm

SoundsSerious, 2024-01-21:
@murrayrm 

Hi, I am currently encountering this problem and still trying to understand exactly what it is doing, I'm gathering it is a convergence check of some kind.

Additionaly i see the note for a more efficient solution and think you might be interested in this dynamics / control library built around openMDAO which is a tool for converging large complex systems. Many of the underlying solvers are still scipy so I think it would fit with the python controls library, but provide better convergence.

https://openmdao.github.io/dymos/index.html