python-control/python-control #462: control.margin()  Floating point versus integer difference

urogers, 2020-10-27:
When running the margin command on Version 0.8.3, the numerator data type impacts the returned results.

Minimum Working Example:   The transfer functions appear identical as expected, but the margin results for the integer are wrong.
_____________________________________
import control as ctrl

num = 50000    # Integer
num2 = 50000.
den = [1, 88., 3840., 25600.]

print(ctrl.margin(ctrl.tf(num,den)))
print(ctrl.margin(ctrl.tf(num2,den)))
___________________________________
Returned Results:
(6.2464, inf, 61.96773353931867, nan)
(6.2464, 101.32058351824355, 61.96773353931867, 13.393128802072123)

PS: Huge fan of this effort.  I have been teaching a 4xx level course in Controls using Python for three years now and truly appreciate the efforts made (Control Library & Slycot)

bnavigator, 2020-10-29:
This apparently has been fixed in the current master branch.

```
greiner@greinerT450s:~/src/python-control/ > python3
Python 3.8.5 (default, Jul 20 2020, 17:41:41) [GCC] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import control as ctrl
>>> num = 500000
>>> num2 = 500000.
>>> den =  [1, 88., 3840., 25600.]
>>> 
>>> print(ctrl.margin(ctrl.tf(num,den)))
(0.6246399999999999, -16.668584884500063, 61.967733539318665, 75.72302661940512)
>>> 
>>> print(ctrl.margin(ctrl.tf(num2,den)))
(0.6246399999999999, -16.668584884500063, 61.967733539318665, 75.72302661940512)
>>> 
>>> print(ctrl.__path__)
['/home/greiner/src/python-control/control']
>>> print(ctrl.__version__)
dev
>>>
greiner@greinerT450s:~/src/python-control/ > git describe  
0.8.3-65-g2d7aad0
```

python-control 0.8.4 is around the corner (#451).

bnavigator, 2020-11-19:
Closing. This is fixed in master and should be released soon.

closed 2020-11-19 by bnavigator